RobotLocomotion/drake #13177: mbp: Support programmatic decomposition into "sub-plants"?

EricCousineau-TRI, 2020-04-29:
## Decomposition

This would cover the solution I mentioned here: https://github.com/RobotLocomotion/drake/issues/12980#issuecomment-614940839
Basically, take an existing mega-plant (e.g. simulation world), and identify sub-graphs, and generate a physically valid `MultibodyPlant` from that.

## Use Cases

* Computing for subsets (as mentioned in #12980)
* Being able to make IIWA controllers without excessive "parameter lumping" (kinematic and inertial)
  * See example in Anzu (can maybe post cut-out of code)
  * See example in `ManipulationStation`: https://github.com/RobotLocomotion/drake/blob/d1e11190794d8b31be5ad5918c8d0f30db7b2874/examples/manipulation_station/manipulation_station.cc#L420-L450

*Meta*: Split off from #13074: https://github.com/RobotLocomotion/drake/issues/13074#issuecomment-614955339

sherm1, 2020-04-30:
I'm not sure I understand what you're asking for, @EricCousineau-TRI. I thought I did from the issue text (extract sub-model from existing model) but then the ManipulationStation example appears to take a full model and then do some post-editing on it after reading it in (adds a couple of welds and a new body). Not clear how those two things fit together -- can you elaborate?

sherm1, 2020-04-30:
Also I'm wondering what are the state transfer needs, if any. Is this useful as just an operation extracting a submodel from a larger one, or must it come with an ability to transfer a portion of the state from an instance of the larger model to an instance of the submodel?

I think we will need to understand the application in some detail in order to fully understand the request. We can discuss at the upcoming Model Mutation Meeting (M³).

EricCousineau-TRI, 2020-04-30:
MATLAB version: https://www.mathworks.com/help/robotics/ref/rigidbodytree.subtree.html

sherm1, 2020-04-30:
Interesting, and not what I would have expected! How would that have helped with the ManipulationStation example you gave above?

Also the documentation there talks only about bodies. Obviously it must copy joints as well, but what about other elements of the multibody system? For example, if there is a spring between two bodies both in the submodel, I think that ought to be copied also. But if one end of the spring is attached to something that is _not_ in the submodel, then that should be ignored I guess. What if there is a weld constraint (or bushing)? That's not a joint so in Matlab would presumably terminate the copying, which seems wrong?

EricCousineau-TRI, 2020-04-30:
> How would that have helped with the ManipulationStation example you gave above?

For that example, one potential workflow that could be enabled by it:
* You have a large scene. It has an IIWA and the gripper.
* If we support deleting / replacing joints (#12703):
  * You clone the IIWA and gripper welded together, then change all gripper joints to be welds.
* If not:
  * You clone the IIWA. Compute the composite inertia of the gripper, then add that body.

That's in contrast to the current workflow:
* You have a large scene, where you've memorized / recorded exactly how you constructed each of them.
* Using your recorded factory info, you create a new IIWA, then add the lumped model.

EricCousineau-TRI, 2020-04-30:
> Also the documentation there talks only about bodies. [...]

Yeah, I dunno. That would definitely be something that needs to spec'd out. However, I think if we nail down a conservative, unambiguous spec for how it should behave (answering your questions and more), then we're good?

EricCousineau-TRI, 2020-05-16:
As mentioned here: https://github.com/RobotLocomotion/drake/issues/13074#issuecomment-629708106

I think my prototype will be handle composition and decomposition in a scoped fashion. When it comes to transferring state, or whatever, it will be done explicitly (e.g. returning associations between plant's bodies, joints, geometries, etc.).

EricCousineau-TRI, 2020-05-19:
Per comment here: https://github.com/RobotLocomotion/drake/issues/13074#issuecomment-630926022

I believe my prototype handles this workflow.