RobotLocomotion/drake #13179: Drake camera not showing texture.

mjm522, 2020-04-30:
Hi,

I am unable to see texture in the Camera view of drake. However it loads correctly in drake visualizer.

[image]

The code that generates the scene is [here](https://github.com/mjm522/doubts)

Is this a known issue? If not, what I could be doing incorrectly?

Best,

Michael

ggould-tri, 2020-04-30:
Sherm, I don't know who currently owns the renderers (@SeanCurtis-TRI ?) so I'm handing this to you to reassign.

SeanCurtis-TRI, 2020-04-30:
I can't 100% explain your problem, but I'm 99% sure the problem comes down to a horrible trap that most people fall into. `drake_visualzier` and the `RenderEngine` interface handle material specifications *very* differently.

  - `RenderEngineVtk` has two ways to associate a color texture with an obj file:
    - takes the obj name (`my_file.obj`) and looks for `my_file.png`. In this case, your obj is named `003_cracker_box_textured_updated.obj` and your texture is named `003_cracker_box_textured.png`. 
    - In the SDF if you assign the tag:
   `<drake:diffuse_map>../meshes/003_cracker_box_texture.png</drake:diffuse_map>`
    that will explicitly inform the render engine that you want *that* texture (of arbitrary name) associated with that obj.
- `drake_visualizer` actually does a passable job respecting the mtl file and is acquiring the texture association that way.

(This is a known issue and once resources free up `RenderEngine` will respect the mtl file as well.)

I have a *big* request. This is a common-enough question that if you could post it in stack overflow with the `drake` tag, others will be able to benefit from your question and this answer.

P.S. Some notes on your code:

- You don't have to declare both `CameraProperties` and `DepthCameraProperties`. If the two sets of properties have identical values for their common properties, then an instance of the latter *is an* instance of the former.
- `frame_id = world_id` only works by accident. `world_id` should be a `BodyIndex` and frame id should be an instance of `FrameId`. What you really want is `frame_id = scene_graph.world_frame_id()`

huihuaTRI, 2020-05-01:
I just want to upvote Sean's suggestion of putting this question to stack overflow. I have fallen into this trap twice :(.

mjm522, 2020-05-01:
Yes this was indeed the thing. Also thanks for the helpful comments about the code. I renamed the files and it worked. It would have been really helpful if it was mentioned in the documentation under the RGBDSensor or CameraProperties class.

SeanCurtis-TRI, 2020-05-14:
To finish resolving this PR, there are two keynotes.

 1. We need to respect the mtl file. There is already an issue for that #11949.
 2. Until that happens, the explanation in this issue needs to be discoverable in the documentation (#13314).

With those two topics covered by other issues, I'm closing this one.

rpoyner-tri, 2022-02-07:
Reopening this, because I pretty much hate omnibus issues that can never be closed (#13314), and don't feel like the original resolution-by-delegation here was really adequate. #16531 will finish the job.

PR #16531 (2022-02-08): [geometry,doc] Clarify RenderEngineVtk texture assignment
Closes: #13179

Also fix a nearby formatting blunder.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16531)
commit b2f00412 (2022-02-08): [geometry,doc] Clarify RenderEngineVtk texture assignment (#16531)

* [geometry,doc] Clarify RenderEngineVtk texture assignment

Closes: #13179

Also fix a nearby formatting blunder.
closed 2022-02-08 by rpoyner-tri