RobotLocomotion/drake #13187: doc: Should add tips on debugging PYTHONPATH issues

EricCousineau-TRI, 2020-04-30:
@mtan3 ran into the following issue here:
```
Traceback (most recent call last):
  File "rviz_demo.py", line 23, in <module>
    from pydrake.geometry import ConnectDrakeVisualizer
ImportError: ImportError: AttributeError: module 'drake' has no attribute 'lcmt_contact_results_for_viz'
```

In the end, it looks like something is messing with her path via site-specific configuration hooks:
https://docs.python.org/3.6/library/site.html

On her machine, what it looks like without telling Python to ignore customizations:
```
$ alias env-isolate='env -i HOME=$HOME DISPLAY=$DISPLAY SHELL=$SHELL TERM=$TERM USER=$USER PATH=/usr/local/bin:/usr/bin:/bin'
$ env-isolate python3 -c 'import sys; print("\n".join(sorted(sys.path)))'
${HOME}
${HOME}/.local/lib/python3.6/site-packages
/usr/lib/python3.6
/usr/lib/python3.6/lib-dynload
/usr/lib/python3/dist-packages
/usr/lib/python36.zip
/usr/local/lib/python3.6/dist-packages
```

The anomaly here is the presence of `${HOME}` on the path. Running Python in isolated mode:
```
$ env-isolate python3 -I -c 'import sys; print("\n".join(sorted(sys.path)))'
/usr/lib/python3.6
/usr/lib/python3.6/lib-dynload
/usr/lib/python3/dist-packages
/usr/lib/python36.zip
/usr/local/lib/python3.6/dist-packages
```

Ideally, we should have some simple tips for debugging stuff like this

jwnimmer-tri, 2020-04-30:
Relates #8475 somewhat.

ggould-tri, 2020-05-01:
Assigning @jwnimmer-tri per related bug, but feel free to reassign as needed.

jwnimmer-tri, 2020-05-01:
=> `component: pydrake` (though Eric hasn't created it yet)

EricCousineau-TRI, 2020-07-20:
Another form of testing environment isolation:
https://gist.github.com/EricCousineau-TRI/dd5bc43eeb05d1fb50277f236872d98b

EricCousineau-TRI, 2020-11-02:
With @hongkai-dai, another form of testing to ensure nothing in the system is messed up:

```sh
env -i \
    DISPLAY=$DISPLAY \
    HOME=$HOME \
    PATH=/usr/local/bin:/usr/bin:/bin \
    SHELL=$SHELL \
    TERM=$TERM \
    USER=$USER \
    bash --norc <<'EOF'
set -eux -o pipefail

cd $(mktemp -d)

env

curl -o drake.tar.gz https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-bionic.tar.gz
tar -xzf drake.tar.gz -C ./ --strip-components=1

python3 -I <<INNER
import sys
sys.path.insert(0, '${PWD}/lib/python3.6/site-packages')
import pydrake.all
print("Done")
INNER
EOF
```

dxyang, 2020-11-05:
I'm trying to follow the steps here and installing drake on Ubuntu 18.04 - https://drake.mit.edu/python_bindings.html#python-bindings-binary. I've run into the same issue with `lcmt_contact_results_for_viz`.

```
(manipulation-veczzm2x-py3.6)  dxy@dxy-ubuntu  ~/manipulation  python test.py
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    from pydrake.multibody.plant import AddMultibodyPlantSceneGraph
ImportError: ImportError: ImportError: AttributeError: module 'drake' has no attribute 'lcmt_contact_results_for_viz'
```

I'm using pyenv + poetry for python management, so I believe this should be a clean, unpolluted copy of python. I unzipped the drake nightly tar file contents into the environment folder and edited my pythonpath to make sure `lib/python3.6/site-packages` from that poetryenv is picked up. However, I get the above error when I run the "example snippet of code from pydrake" from the drake page.

Here are some of the commands referenced earlier in this issue if they help with debugging. Happy to move this to a new issue if needed. 

```
(manipulation-veczzm2x-py3.6)  dxy@dxy-ubuntu  ~/manipulation  which python
/home/dxy/.cache/pypoetry/virtualenvs/manipulation-veczzm2x-py3.6/bin/python

(manipulation-veczzm2x-py3.6)  ✘ dxy@dxy-ubuntu  ~/manipulation  python -c 'import sys; print("\n".join(sorted(sys.path)))' 
/home/dxy/.cache/pypoetry/virtualenvs/manipulation-veczzm2x-py3.6/lib/python3.6/site-packages
/home/dxy/.pyenv/versions/3.6.9/lib/python3.6
/home/dxy/.pyenv/versions/3.6.9/lib/python3.6/lib-dynload
/home/dxy/.pyenv/versions/3.6.9/lib/python36.zip
/home/dxy/manipulation
```

jwnimmer-tri, 2020-11-05:
When the error message is "module 'drake' has no attribute...", the investigation should be `import drake; print(drake.__file__)` to see where the drake module is coming from.  It should be some "lib/python3.6/site-packages" folder where Drake is installed, not a Drake source tree.

Here's an example using the precompiled Drake binaries:
```console
jwnimmer@call-cps:~$ env PYTHONPATH=~/Downloads/v0.24.0/drake-20201026-bionic/drake/lib/python3.6/site-packages python3 -c 'import drake; print(drake.__file__)'
/home/jwnimmer/Downloads/v0.24.0/drake-20201026-bionic/drake/lib/python3.6/site-packages/drake/__init__.py
```

dxyang, 2020-11-05:
Ahh thanks! I had a copy of the repo checked out that was causing those import issues.

closed 2024-03-21 by jwnimmer-tri