RobotLocomotion/drake #13202: doc: Minimum MacOS Version = 10.13

aiyer-commits, 2020-05-01:
Hi Team,

Given that the minimum XCode version required is XCode 10, I think the docs should explicitly state that MacOS 10.13 is required to use drake/pydrake. This is because the last XCode version for MacOS 10.12.6 is XCode 9.2.

Just a heads up!

jwnimmer-tri, 2020-05-01:
We have some information here https://drake.mit.edu/developers.html#supported-configurations that seems to specify this already?  But perhaps it's not linked from enough places.  Where in the docs did you try to find the answer.  We can perhaps add a link from there to this table.

aiyer-commits, 2020-05-01:
Oh whoops, I didn't check that page. I was simply going through the installation pages (https://drake.mit.edu/python_bindings.html#python-bindings-binary)

I think adding "10.14" on this page to match the format for Ubuntu ("18.04") would make it totally explicit here:

[image]

jwnimmer-tri, 2020-05-01:
Great point, we'll work on it.

aiyer-commits, 2020-05-01:
Also, maybe this issue should be renamed since I didn't actually test if drake can be installd on 10.13... whoopsie!

jamiesnape, 2020-05-04:
> I think adding "10.14" on this page to match the format for Ubuntu ("18.04") would make it totally explicit here:

So the point is that the version does not matter for the example and I don't think we want to duplicate "Mojave and Catalina" everywhere. The Bionic example would be incorrect in Focal, so it is qualified.

aiyer-commits, 2020-05-04:
Oh I didn’t realize that Focal would require a different set up from Bionic! That makes sense to me. 

My suggestion came from the perspective of being impatient and skipping the “working configurations” section of the docs haha