RobotLocomotion/drake #13210: doc: Specify Ubuntu Desktop Edition Preferred

aiyer-commits, 2020-05-02:
Hi Team,

I've been experimenting with running drake in Virtualbox, running Ubuntu 18.04 Server Edition (which does not come with X11 installed), and only realized that XOrg/X11 is needed while running the "rendering multi-body plant" tutorial via jupyter notebook. I realized this thanks to vtk crashing my jupyter notebook kernel, as seen in this screenshot:

[image]

I am working on configuring my ubuntu server edition vm to use X11, will report back on if it can be done.

aiyer-commits, 2020-05-02:
Ok I got it to work in Ubuntu Server Edition by installing xorg and xvfb, following the instructions here (https://docs.pyvista.org/getting-started/installation.html). 

Would it make sense to add docs specific to Ubuntu Server Edition?

jwnimmer-tri, 2020-05-04:
Yes, when listing Drake's dependencies it probably makes sense to note that any rendering-related features require an X server.

jamiesnape, 2020-05-04:
Alluding to the title, I don't think we (should) prefer either. CI uses "minimal" FWIW.

jwnimmer-tri, 2020-05-04:
Good point. I'd meant to update the title during triage, but forgot.