RobotLocomotion/drake #13212: Build error on Ubuntu 18.04

ivandariojr, 2020-05-03:
After following the environment installation instructions for Ubuntu 18.04 I encounter the following error when running `bazel build //...`: 

```
project/drake/systems/framework/BUILD.bazel:959:1: C++ compilation of rule '//systems/framework:system_constraint_test' failed (Exit 1) gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g '-std=c++0x' -MD -MF ... (remaining 165 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from external/eigen/include/_usr_local_include_eigen3/Eigen/Core:19:0,
                 from bazel-out/k8-dbg/bin/common/_virtual_includes/autodiff/drake/common/autodiff.h:12,
                 from bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/system_constraint/drake/systems/framework/system_constraint.h:9,
                 from systems/framework/test/system_constraint_test.cc:1:
external/eigen/include/_usr_local_include_eigen3/Eigen/src/Core/util/Macros.h: In instantiation of 'bool Eigen::internal::all(T, Ts ...) [with T = Eigen::CwiseBinaryOp<Eigen::internal::scalar_cmp_op<double, double, (Eigen::internal::ComparisonName)0>, const Eigen::ArrayWrapper<Eigen::Matrix<double, -1, 1> >, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Array<double, -1, 1> > >; Ts = {}]':
systems/framework/system_constraint.h:223:19:   required from 'drake::boolean<T> drake::systems::SystemConstraint<T>::CheckSatisfied(const drake::systems::Context<T>&, double) const [with T = double; drake::boolean<T> = bool]'
systems/framework/test/system_constraint_test.cc:115:3:   required from here
external/eigen/include/_usr_local_include_eigen3/Eigen/src/Core/util/Macros.h:1089:36: error: no match for 'operator&&' (operand types are 'Eigen::CwiseBinaryOp<Eigen::internal::scalar_cmp_op<double, double, (Eigen::internal::ComparisonName)0>, const Eigen::ArrayWrapper<Eigen::Matrix<double, -1, 1> >, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Array<double, -1, 1> > >' and 'bool')
 bool all(T t, Ts ... ts){ return t && all(ts...); }
                                  ~~^~~~~~~~~~~~~
```

Here is some relevant system information:

 - **Operating system**: Ubuntu 18.04
 - **Language** : C++
 - **C++ compiler**:
```
g++ --version                                     
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
```
 - **Python version**:
```
python3 --version
Python 3.6.9
```
- **Python distribution**: 
apt

- **If building from source**:

 - **Build system:** 

 - Bazel and CMake (I tried on both and got the same error)
 - Bazel version: 3.1.0 (from APT) and 3.0.0 (as installed through install_prereqs.sh)
   I tried both.
```
which bazel    
/usr/bin/bazel
```     
- **Bazel C++ compiler**

```
bazel run @drake//common:print_host_settings
INFO: Analyzed target //common:print_host_settings (20 packages loaded, 86 targets configured).
INFO: Found 1 target...
Target //tools/cc_toolchain:print_host_settings up-to-date:
  bazel-bin/tools/cc_toolchain/print_host_settings
INFO: Elapsed time: 3.187s, Critical Path: 0.10s
INFO: 1 process: 1 linux-sandbox.
INFO: Build completed successfully, 3 total actions
INFO: Build completed successfully, 3 total actions
++ uname -s
+ [[ Linux == Darwin ]]
+ capture_cc_env=tools/cc_toolchain/capture_cc.env
+ source tools/cc_toolchain/capture_cc.env
++ BAZEL_CC=/usr/bin/gcc
++ BAZEL_CC_FLAGS=
+ [[ ! -z /usr/bin/gcc ]]
++ python3 -c 'import os; print(os.path.realpath("/usr/bin/gcc"))'
+ BAZEL_CC=/usr/bin/x86_64-linux-gnu-gcc-7
+ /usr/bin/x86_64-linux-gnu-gcc-7 --version
x86_64-linux-gnu-gcc-7 (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
```
- **CMake version** (which cmake; cmake --version)
```
/usr/bin/cmake
cmake --version                                                     
cmake version 3.10.2
```
 - **CMake C++ compiler** (cmake -LA <path_to_source_dir> | grep 'CMAKE_.*_COMPILER')
```
cmake -LA . | grep 'CMAKE_.*_COMPILER'
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++
CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-7
CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-7
CMAKE_C_COMPIfunctionLER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-7
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-7
```
 - **Git revision** (git rev-parse --short HEAD) master branch

```
git rev-parse --short HEAD   
69c771fff
```
- **Eigen Version** : I installed it through apt but just in case it's relevant: 3.3.9

jwnimmer-tri, 2020-05-03:
The presence of `_usr_local_include_eigen3` in the error message looks like it's using a copy of Eigen installed into `/usr/local/include`, not the Ubuntu version of Eigen from `/usr/include`.  Can you confirm that there's a copy of Eigen in `/usr/local/include`, and what version it is?

The Ubuntu version of Eigen is 3.3.4 (https://packages.ubuntu.com/bionic/libeigen3-dev), I'm not sure why you say you installed 3.3.9 via apt?

ivandariojr, 2020-05-03:
Thank you. Something was bogus about the eigen version I had installed. After downgrading to 3.3.4 the problem went away.

closed 2020-05-03 by ivandariojr