RobotLocomotion/drake #13214: Support contact results visualization in PlanarSceneGraphVisualizer

RussTedrake, 2020-05-03:
This is one simple feature that we're missing in the jupyter (e.g. colab/binder) workflow.  We have it for `drake_visualizer` (with `ConnectContactResultsToDrakeVisualizer`) and for `meshcat`.

kamiradi, 2024-02-28:
Hi, I have modified the intial `PlanarSceneGraphVisualizer` to now display contact points. As per our discussion [here](https://stackoverflow.com/questions/77908362/is-there-a-way-to-visualise-contact-forces-in-2d-in-drake/77909220?noredirect=1#comment137389672_77909220), I wanted to see if I could contribute this to the code base. 

Would love your feedback on the code as well!

RussTedrake, 2024-02-29:
That sounds fantastic.  Please do open a PR.  I think @bernhardpg might be particular interested in seeing it land.

kamiradi, 2024-02-29:
See #21074.