RobotLocomotion/drake #13215: drake::log() entries are invisible on colab/binder

RussTedrake, 2020-05-04:
Can we make the sdplog output appear in the jupyter window instead of in the terminal?  I've seen a few users very confused by something that would have been obvious if they had seen the console warning.

EricCousineau-TRI, 2020-06-04:
I think `pybind11` has an awesome mechanism for this that we could use:
https://pybind11.readthedocs.io/en/stable/advanced/pycpp/utilities.html#capturing-standard-output-from-ostream

For both `stdout` and `stderr`

EricCousineau-TRI, 2020-06-04:
I think this is also effectively a `pydrake` issue, just b/c streams should be synchronized. Adding both tags for now.

EricCousineau-TRI, 2020-06-04:
Solution also should not be Jupyter-specific, it should just be on for `pydrake`; edge cases may crop up later (e.g. running Python stuff from C++ console); we can address them then.

EricCousineau-TRI, 2020-06-04:
@BetsyMcPhail I think this may only take about ~5 Perfect Engineering hours to tackle.
Is it possible for you to try this out, ahead of the `mkdoc` docstring formatting issue, within the next 2 weeks?

jwnimmer-tri, 2020-06-04:
@EricCousineau-TRI can you clarify how you'd like to see the tools on that page (#capturing-standard-output-from-ostream) used?  I don't think per-function nor context-manager use cases meet the need.  Are you suggesting that we invoke the same underlying feature, but in a way that happens upon pydrake import and never turns off?

I'm not sure that we should redirect all of cout & cerr either?  That seems a bit much for a library to take over the whole application.  Redirecting just `drake::log` seems better, and should be done by replacing the spdlog sink.

BetsyMcPhail, 2020-06-08:
> @BetsyMcPhail I think this may only take about ~5 Perfect Engineering hours to tackle.
> Is it possible for you to try this out, ahead of the `mkdoc` docstring formatting issue, within the next 2 weeks?

Sounds good

commit 112cb9b4 (2020-07-23): Provide a spdlog sink that connects to Python's `logging` module (#13568)

* Provide a spdlog sink that connects to Python's `logging` module

sdplog output will appear in the Jupyter window instead of in the terminal.

Add tests for python logging redirection

Resolves #13215
closed 2020-07-23 by BetsyMcPhail