RobotLocomotion/drake #13219: Build IBEX from source using Bazel

jamiesnape, 2020-05-04:
Now dReal is built from source there is less need to use a system IBEX, which is prone to breakage on macOS when its Homebrew dependencies change. Not completely trivial, but not difficult at all to do.

FYI @soonho-tri.

jamiesnape, 2021-04-29:
Most likely a prerequisite for #1183.

jwnimmer-tri, 2021-09-25:
As a reminder, because IBEX is licensed as LGPL, our BUILD file for it must create _only_ a shared library.  Refer to `drake/tools/workspace/lcm` for an example.

RussTedrake, 2021-10-01:
I've briefly started looking into this, as discussed in #15789.  Will update this issue if we decide to move forward.

soonhokong, 2021-10-02:
Let me enumerate some findings regarding this topic here first:

 - IBEX dropped CLP support starting from `2.8.7`. The other supported LP solvers are [SoPlex](https://soplex.zib.de/) and [CPLEX](https://www.ibm.com/analytics/cplex-optimizer). CPLEX is proprietary. SoPlex is released under the [ZIB Academic License](https://scipopt.org/academic.txt). It's free only if you're a member of a *non*-commercial company or academic institute.
 - There is an attempt to bring CLP back to IBEX. See https://github.com/ibex-team/ibex-lib/discussions/489. Note that this task is not completed yet.
 - IBEX is using the [waf](https://waf.io/) build system. I think this is one of the pain points in building IBEX from source using bazel. There is an ongoing effort to support CMake. For example, [ibex-2.8.6](https://github.com/ibex-team/ibex-lib/tree/ibex-2.8.6) includes `CMakeLists.txt` file. However this is still experimental.

My Action Item:
 - Check IBEX-2.8.6 + CLP if it can resolve #15789.

A question: Can we have Kitware's time to cmake-ify ibex?

RussTedrake, 2021-10-02:
>    ibex-team/ibex-lib#489

I suspect I could get the bazel build working with a few more hours of
work. Should I target 2.8.6? Is there any constraint coming from dReal that
might hold the version back?

soonhokong, 2021-10-02:
There are some patches that we need to re-apply including the patch I made to make filibxx (interval lib) C++17 compatible. Also, we need to update Drake's IBEXSolver implementation as they changed some of the configuration structure(s).