RobotLocomotion/drake #13231: Remove ruby dependency

jwnimmer-tri, 2020-05-04:
Asking users to install a full ruby interpreter just to build Drake from source seems overly heavyweight.

We only use it for https://github.com/osrf/sdformat/blob/master/sdf/embedSdf.rb, so it should be possible to rewrite that script in a lighter language (either within Drake, or upstream).

See #8976 for where we added ruby.

jwnimmer-tri, 2020-05-05:
My current plan of attack:
- PR to sdformat refactoring the "embed" codegen logic
  - remove application-layer versioning logic from codegen
    - be a boring dictionary of relative_filename => contents
      - the codegen rb script will be like 10 lines long
  - stop relying on the global initialization and destruction fiasco
- Upgrade sdformat in Drake
- Re-implement the (now boring) codegen script in python

EricCousineau-TRI, 2020-05-05:
FTR See old BitBucket PR discussion:
https://github.com/osrf/sdformat/blame/9eb6b99c2c8295fcf533589b4e8e6f4fd5fa623a/Changelog.md#L527
https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/434/page/1#comment-65236228

jwnimmer-tri, 2020-05-07:
We'll see where https://github.com/osrf/sdformat/pull/270 takes us.

PR #13262 (2020-05-08): workspace: Remove dependency on ruby
Cherry-pick a patch from sdformat upstream to make the data embedding easier, and then re-implement the embedder in python.

Deprecate the `@ruby` external.

Closes #13231.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13262)
closed 2020-05-08 by jwnimmer-tri