RobotLocomotion/drake #13241: [geometry] Decompose QueryObject into multiple QueryObject flavors.

SeanCurtis-TRI, 2020-05-05:
# Current state

Currently all queries, of any type, on the state of geometry in the world pass through a single class: `QueryObject`. This includes asking what geometries are in the world, what their pose is, what color they are, if they intersect, what they look like in a camera, etc.

# Problem

This monolithic query object leads to several issues:

  - The ability to make *any* query on the `QueryObject` includes a cost that prepares *all* queries for evaluation.
    - Although this *can* be separated under the monolithic `QueryObject` it would be greatly facilitated if the wholly orthogonal pieces (e.g., `ProximityEngine` vs `RenderEngine`) were segregated structurally.
  - The monolithic query can lead to algebraic loops.
    - As a not-quite-hypothetical, [imagine a system](https://github.com/SeanCurtis-TRI/drake/blob/PR_painting_meshes/geometry/mesh_painter_system.h) that creates data that rendering depends on, but to create that data, the system in turn depends on geometry poses. So, it connects an input port to a `QueryObject` to acquire geometry pose information and outputs its data. That data, because rendering depends on it, will eventually loop around to input to SceneGraph and the monolithic `QueryObject` will depend on it. An algebraic loop is introduced.
    - However, if the system can declare that it only cares about poses in a more targeted way, *excluding* the possibility of making render queries, the loop is broken.

# Proposed Solution

In short, we break up the `QueryObject` into a family of query objects, segregating the queries. We have defined geometry "roles" and it seems reasonable that the decomposition would follow similar lines. So, we'd have the following `QueryObject`s (replacing the monolithic `QueryObject`):

  - `PoseQueryObject`: Gives access to:
    - a `SceneGraphInspector` (to introspect all aspects of geometry that *don't* depend frame pose).
    - poses of frames and geometries that *do* depend on frame poses.
  - `ProximityQueryObject`: Gives access to:
    - everything that `PoseQueryObject` provides
    - contact queries
    - distance queries
  - `PerceptionQueryObject`: Gives access to:
    - everything that `PoseQueryObject` provides
    - render queries (depth, color, label rendering)

Note: giving `PerceptionQO` and `ProximityQO` the `PoseQueryObject` functionality isn't *strictly* necessary. However, intuition suggests as long as you're dependent on either proximity or render engines (which themselves depend on the poses of geometries in the world), it seems reasonable to also expose those queries directly.

# Proposed implementation
 
  - `PoseQueryObject` would be the new base class for the family.
    - In addition to the queries outlined above, it would also uniquely implement the "live" and "baked" `QueryObject` functionality which makes `QueryObject` `AbstractValue`-compatible.
  - `Proximity\PercetionQO` would derive from `PoseQueryObject`
    - the relationship would *not* be polymorphic. 
    - The children classes provide *no* new state, they simply give access to new queries that are implemented elsewhere.
  - Each of the new `P*QueryObject` types will correspond to a new port in `SceneGraph`.
  - The old port and old `QueryObject` class will be deprecated.

SeanCurtis-TRI, 2020-05-05:
cc @jwnimmer-tri @ggould-tri

jwnimmer-tri, 2020-05-05:
I still remain extremely skeptical about implementation inheritance, especially since all of these 3QOs would still undergo type erasure.  I think that without understanding how GeometryState gets split into pieces so that the 3QOs each use some of it (+ its associated cache entries, lookup tables, etc), we can't responsibly reason about whether or not the QOs will be a valid use case for implementation inheritance.

It seems trivial to use ProximityQO has-a PoseQO instead of is-a PoseQO, and avoid inheritance entirely.  Can you outline any end-user code snippets or implementation difficulties that would arise from using the boring answer (has-a) instead of the brittle answer (is-a)?

SeanCurtis-TRI, 2020-05-05:
> understanding how GeometryState gets split into pieces

This is an assumption that is not related to this issue. None of this requires `GeometryState` to change. If it changes, it won't have anything to do with this change.

However, I appreciate the point that public inheritance may imply more to the API than it should. There are a lot of unknowns. Private inheritance would get the desired effect with the minimum of overhead without creating an API that would be difficult to unwrap later. How would you feel about that?

jwnimmer-tri, 2020-05-05:
_"When using inheritance, make it public."_
-- https://drake.mit.edu/styleguide/cppguide.html#Inheritance

I'm not sure what private inheritance buys the user vs has-a, anyway?  In either case, they have to call `proximity_query.poses().X_FM(id)` to look up a pose instead of just `proximity_query.X_FM(id)`.

And inside the implementation, `private: /* member */ PoseQO qo_;` vs `ProxQO : private PoseQO` are a wash?

SeanCurtis-TRI, 2020-05-05:
My primary reason for inheritance is less about the public API and more about the under-the-hood live/bake functionality.

But I can go ahead and wrap that into its own class that each of the QOs has.

SeanCurtis-TRI, 2020-05-05:
I'm also not a big fan of one QO "has a"-ing another QO. So, I'll probably refactor shared functionality into something that *can* be has-a'd.

jwnimmer-tri, 2020-05-05:
I have no opinion on whether or not the private data has-a `PoseQueryObject` or some helper type instead.

The API question is whether there be a getter to fetch a `PoseQueryObject` from one of the other souped-up query objects, as in `proximity_query_object.poses()`.

Imagine that a user has a method named `MySystem<T>::CalcFoo` that makes proximity queries.  Clearly the `MySystem` has an input port for the `ProximityQueryObject`.  If `CalcFoo` also needs to pass a `PoseQueryObject` into a utility function that only know about poses (and not proximity), can `CalcFoo` obtain the `PoseQueryObject` from the `ProximityQueryObject` directly, or does the user need to wire up a distinct `pose_query` input port into `MySystem`?

jwnimmer-tri, 2020-05-06:
(To be clear: I don't actually know the right answer.  I assume examples will guide us.  But I think its the right question to ask for API design.)

SeanCurtis-TRI, 2020-05-06:
Ha. I was thinking about the subject of your last posting this morning while walking my dogs. And then I came in to share my thoughts and found that you'd been thinking about it as well.

Here are my thoughts:

1. If there is some kind of "common" query across QOs, I don't like the idea that the spelling changes.  I.e., `PoseQueryObject:` has `query_object.X_WG()` vs `ProximityQueryObject` with `query_object.poses().X_WG()`.
  - Two options for avoiding this
    - Inheritance - the more I think about inheritance, and the relationship between QO, SceneGraph (which owns all of the caches) and GeometryState (which owns the operations), the more I'm convinced that inheritance is a viable way forward. 
    - Wrapping the "has-a" thing with a local API. In my estimation, this is overly brittle.

2. If I can't have the same spelling of common queries via some reasonable mechanism, then I'd rather forego common queries.
  - It may well be that perception of the need for common queries is erroneous. If pressed, I can't give a use case beyond theoretical ones -- with one exception, the ability to get a `SceneGraphInspector`  is the same for all -- in any context, you should always be able to ask about geometry properties. So, all QO flavors would need to have an inspector -- they can do that independently without too much overhead (although, again, more conveniently addressed via inheritance).
  - The implication is, yes, if you want poses *and* proximity, you need to connect to two ports.

3. As for the idea of accessing one query object via another. I hadn't considered it, it's doable but inconvenient. And, at first glance, I'm not in favor of it.
  - It would have to be limited. Given QO types A, B, and C, B and C can get you an A, but not vice versa. That asymmetry would have to be documented to make it clear that they shouldn't expect symmetry. I'd rather not incur that documentation burden without shown cause.

sherm1, 2020-05-06:
I rather like the idea of having several ports with disjoint functionality. That seems simplest to code and explain, and seems cleaner in terms of the declared port prerequisites. Is that too awkward? MBP has lots of ports like that for various bits of information and I'm not aware of any complaints about that.

SG could have Inspector, Pose, Proximity, Perception etc. ports with no duplicate functionality. Maybe deprecation would be easier also since the original Query port could coexist for a while?

SeanCurtis-TRI, 2020-05-06:
If we want to go all in on separated functionality (i.e., including `SceneGraphInspector`), I can get behind that for all of the reasons enumerated.

jwnimmer-tri, 2020-05-11:
We are blessed with many examples already of different kinds of call sites.  The easiest and best way to assess the impact to users is to prototype a port of Drake (and TRI Anzu) to the new APIs.  Then we can look at concrete examples of using the hypothetical new APIs.

SeanCurtis-TRI, 2021-12-04:
Closing as probably won't do.

closed 2021-12-04 by SeanCurtis-TRI