RobotLocomotion/drake #13242: Creating an example to model and control mobile manipulator such as PR2

huihuaTRI, 2020-05-05:
I have learned several useful modeling skills when building a mobile-based robot. I would like to transfer that experience to Drake as an example for general users. Additionally, Drake does not have an example of a mobile-based robot yet. This effort will fill this vacancy.

The original plan is to use HSR as the example model. However, due to the mesh files of HSR is not open-source friendly, we have to switch to use PR2. The code will be designed to be generic. In the future, an HSR model can be easily added if the meshes files can be truly open-sourced.

The initial work of the HSR simulation can be found in https://github.com/huihuaTRI/drake/tree/021920_hsr_sim_stub/examples/hsr.

The example will include three parts
1. A main SimWorld diagram system that includes the robots and the environment. This part is very similar to the [manipulation_station](https://github.com/RobotLocomotion/drake/tree/master/examples/manipulation_station) example in Drake.
2. A robot parameter loader to load the robot and world setups from yaml file. A common robot parameter definition allows the possibility to design controllers in a more generic way.
3. A main controller that fuses different controllers for different parts together. Two simple controllers will be provided to show as an example.

huihuaTRI, 2020-05-08:
Hi @jwnimmer-tri, could you please take a look at this branch: https://github.com/huihuaTRI/drake/tree/050720_pr2_tutorial_idea2.

The comments are not updated yet. I want to confirm with you that this is the right direction before I polish the code. Thanks.

jwnimmer-tri, 2020-05-14:
The README says

>  This simulates a PR2 holding the zero position using two controllers for different parts of the robot.

Is there a plan to have anything move?  Either a fixed plan, or teleop, or ...?

huihuaTRI, 2020-05-14:
The original plan is to use an IK to compute a position. The code I pushed is what I currently have for the HSR simulation. The IK is not ready yet.

jwnimmer-tri, 2020-05-19:
Maybe #13325 has some reusable IK help.

huihuaTRI, 2020-05-19:
Thanks, Jeremy. I actually want to get your confirmation on whether what I currently have aligns with the preferred way of adding a tutorial to Drake.

jwnimmer-tri, 2020-07-13:
There is a lot of code infrastructure for easy-to-configure Kp gains & etc, but if the robot and sim don't move anyway, it seems like a lost of noise with little conveyable value to a consumer of this example.  If the key point is just how to deal with wheels, it seems like less code could get the job done.

huihuaTRI, 2020-07-20:
I have the same feeling that this work may not bring enough merit to Drake examples. I will have to replan about how to move this effort forward if necessary.

sherm1, 2022-04-26:
Would be great to have an example like this but I'm going to close the issue now since it's been a couple of years.

closed 2022-04-26 by sherm1