RobotLocomotion/drake #13244: styleguide: Should support nested namespaces?

EricCousineau-TRI, 2020-05-06:
Came up in:
https://github.com/robotlocomotion/drake/pull/13225#pullrequestreview-406269679

Seems like current upstream styleguide still doesn't acknowledge nested namespaces either:
https://google.github.io/styleguide/cppguide.html#Namespace_Formatting

FTR nested namespaces make:
```cc
namespace drake::pydrake {
...
}  // namespace drake::pydrake
```
equivalent to:
```
namespace drake {
namespace pydrake {
...
}  // namespace pydrake
}  // namespace drake
```
https://en.cppreference.com/w/cpp/17

sherm1, 2020-05-06:
We definitely should allow that! Much nicer.

avalenzu, 2020-05-06:
TIL!

jwnimmer-tri, 2020-05-06:
I am not a huge fan of "there is more than one way to do it", for all the usual reasons.  If we like the more compact style, then we should plan to update style guide to disprefer multiple lines and reformat all of Drake in the new style.  Having a linter rule would be a bonus.

EricCousineau-TRI, 2020-05-06:
SGTM. Was thinking there should be a list of different options, with preferences.

jamiesnape, 2020-05-11:
Is this a question for a feature request?

jwnimmer-tri, 2020-05-14:
I think it was both, but any case is on Eric's plate to develop the linter if he want to change the style.

jwnimmer-tri, 2024-11-01:
The idea is a fine one, but we don't need an issue tracking it.

closed 2024-11-01 by jwnimmer-tri