RobotLocomotion/drake #13247: Users need to update Parameters during simulation

sherm1, 2020-05-06:
(This is a partial follow-up from the Mutating Multibody Models Meeting ([notes](https://docs.google.com/document/d/1a4y_Zgph5gUqPoZrnln_B7qfteFP4TOhPteqr8BpAQk)) and relates to the epic #13073.)

### User request
One of several high-priority requests was to be able to modify model parameters during a simulation (mass, contact, and friction properties were mentioned specifically).  This is a two-part request:
1. our models need to look for these values in the mutable Context, not the const System, and
2. there needs to be a way to modify those values as a result of events in the simulation.

### Discussion
The obvious first thought is to do this with Drake's Parameter feature. However, as pointed out in #7017 Parameters have some notable weaknesses. Most relevant for this discussion is the fact that there is no Drake event handler capable of updating Parameters (the so-called `UnrestrictedUpdate` 
 handler permits modification of only State). The only available way to update is to stop a simulation entirely, modify Parameters, then restart. We would need to add a new handler type `ReallyUnrestrictedUpdate` that would permit modification of Parameters as well as state.

It is worth noting that the concept "parameter" adds nothing that is not already including in the concept of "discrete state" (including abstract-valued discrete state). Modifiable parameters are simply discrete state variables that don't get updated very often.

### Proposal
Reimplement Parameters using discrete (and abstract) state rather than as a separate entity. (We could retain some sugar if necessary.) That would resolve #7017 and would permit the existing event handlers to modify parameters as desired (request 2 above). This is also consistent with the current proposal for reworking Drake's variables (see #9171 and [this proposal](https://docs.google.com/document/d/1cTgWeuSqSRg5XyFujzx8ZGdmF9W_3wpp58bQ1xq2zaE/edit#heading=h.2mkpwoq97rh3)).

To address request 1 above, any LeafSystem with modifiable parameters (particularly MBP) would declare discrete or abstract state to hold the runtime values of those parameters. The current construction-time values would become the default values for those state variables. That is a series of separate projects, mostly independent.

Thoughts?

cc @edrumwri @naveenoid

jwnimmer-tri, 2020-05-06:
\CC @RussTedrake for sure, since this "Parameters get copied into State" affects our Systems model of problems.

Sherm:
> ... to modify model parameters during a simulation (mass, contact, and friction properties were mentioned specifically) ... Drake event handler capable of updating Parameters  ...

I'll grant that _in general_ event-based updates to the quantities you listed are useful, but for the actual use cases that those users wanted, is it a strict requirement that the update come from within the diagram?  (The alternative is that code above the simulator -- the application layer -- calls setter methods on a System a Context to change parameters.)  Can you talk us through an example situation where, e.g., friction properties have to change within the diagram?  How does the System deciding the need for a change influence a sibling system without an output-input relation?  A subsystem event cannot change the discrete state of a sibling system, only its own state.  And we cannot add extrinsic update events to a Diagram yet that I'm aware.

I'm wondering if a simple solution to the actual user need is at hand, versus what sounds to me like a complicated new message-passing system between systems.

sherm1, 2020-05-07:
Those are good questions. I _believe_ the use case would almost always be that an event detected by MBP would be handled within MBP and would modify parameters owned by MBP. I don't think it requires one subsystem updating another (except through ports as usual).

We don't currently have a mechanism for detecting an event in a subsystem and interrupting the whole simulation (#4447, #10445) although the Simulator provides a monitor() method with that capability (and it wouldn't be hard to provide more general simulation interruption).

But this is really a question for the requesters: @naveenoid, @edrumwri if there were suitable parameters available would you need to modify them in an event handler or would it be sufficient to interrupt the simulation, modify parameters, then restart the simulation from there?

jwnimmer-tri, 2020-05-07:
Huh.  I did not anticipate that any plausible use case was fully and autonomously encapsulated within MbP!  I'd love to hear those user stories.

But in any case, if MbP is changing its masses over the course of a simulation, then for sure that is MbP State and not MbP Parameters -- you have my full agreement there.

We just have to be careful not add it as State unless the user wants such a modeling choice. Naive users should not have to pay for the analysis burden of extra declared State, c.f., model hierarchies (#6927) and relatedly #9501.

sherm1, 2020-05-07:
My point is that there is no actual difference between "parameters" and "discrete state" in Drake. There might be a difference in the user's intent, though even that is blurry if "parameters" are being updated during a simulation (e.g. Naveen wants to modify friction coefficients after the robot cleans a dirty surface). I believe Russ has some narrow situations in mind for analysis (like discrete systems governed entirely by difference equations) but Drake admits a much broader class of systems.

The real distinctions among the various model types must be obtained by introspection of the variables and the update methods, but the current architecture can't capture the necessary relationships. The [State 2.0 proposal](https://docs.google.com/document/d/1cTgWeuSqSRg5XyFujzx8ZGdmF9W_3wpp58bQ1xq2zaE/edit#heading=h.2mkpwoq97rh3) fixes that but requires major surgery.

We could continue to permit some variables to be designated "parameters" if this conveys some important distinction among the variables. I don't currently see what is that distinction? And this is only the mildest of the "model mutation" requests -- next up is being able to add and remove bodies (during simulation) which would change the number of state variables and port sizes.

RussTedrake, 2020-05-07:
If there was no actual difference between “parameters” and “discrete state” in Drake, then we wouldn’t have this issue.  Parameters are const over the lifetime of a simulation — that is an extremely important concept.  As you and Jeremy both say, I don’t want to clutter my state space (for control design, analysis, even serialization of a Context, if we get there) with big lumps of parameters that are simply constant.
Having the parameters share the tooling for discrete state is fine with me, but is almost orthogonal, I think.

I think I would be ok with saying that unrestricted update events (even without adding a new type for ReallUnrestrictedUpdateEvent) are allowed to update the Parameters.  The name already implies to me and any algorithm writer that they should not expect simple things to happen for a system that has these events registered.

Otherwise, I would think that MBP, SceneGraph, or other systems that want to have a mode that enables mutable parameters should store their parameters as discrete state instead of parameters if that option is enabled.

jwnimmer-tri, 2020-05-07:
The key insight of the State 2.0 proposal (wrt this thread) is that we'd have per-state-group update handlers, instead of allowing handlers to update anything in their entire category. For, e.g., control design, the State vs Parameter (as we term them currently) would come down to whether a given state group has any update mechanism (aka "time advancement strategy") defined on it.  If not, it is const.

> We could continue to permit some variables to be designated "parameters" if this conveys some important distinction among the variables. I don't currently see what is that distinction?

The important take-away is the const-ness (i.e., those without any updates / derivatives).  Once we have a State 2.0 mechanism to ask of a state group is const, then we can ditch the separate concept of Parameters and just use the more nuanced state-interrogation methods instead.  But until that point, we need to be careful to avoid storing const-state into Context State in the meantime.

(It seems like we have general agreement on the above, I'm just trying to restate it to make sure.)

I'm not sure if I agree with Russ that unrestricted update events should be able to change the Parameters (until State 2.0 comes along), though I do see how it could help meet the goals from the top post, and maybe its the right compromise.  Perhaps debating the specifics with some concrete examples would help.

In any case, I'm still super curious to hear from the requestors whether intra-plant events are even relevant in the first place, or if it's instead just as simple as adding a `set_mass` to the MbP API and calling it a day.

amcastro-tri, 2020-05-07:
I believe I am in agreement with y'all on how to handle parameters and I love where the design discussion is going.
However, I want to be very specific about the discussion we had on [MBP/SG mutation](https://docs.google.com/document/d/1a4y_Zgph5gUqPoZrnln_B7qfteFP4TOhPteqr8BpAQk/edit); in particular in answer to @jwnimmer-tri's question:

> I'll grant that in general event-based updates to the quantities you listed are useful, but for the actual use cases that those users wanted, is it a strict requirement that the update come from [...]

Yes, for most (if not all really) the actual need from the users was that the application layer would do some sim; stop; call setters; continue sim with new params. That, if within an application layer doing sim. There was the use case of controller Diagrams not within  a sim layer. 
Therefore, in concrete connection with what was discussed in that meeting, the event handling of parameters was definitely a far future nice to have, but not the actual request.

Regarding the **continuous** change of contact properties for @naveenoid's application, I am 100% sure that is **not about parameters** but about introducing a new model, possibly even with its own state. We can discuss separately in another channel if you feel confused about my statement. His application really calls for a new physical model (even if a simple algebraic one), not parameters.

jwnimmer-tri, 2020-05-07:
> Therefore, in concrete connection with what was discussed in that meeting, the event handling of parameters was definitely a far future nice to have, but not the actual request.

@amcastro-tri Great! Therefore, for the "simple C++ setters" APIs that you actually want, can you speak to this statement from Sherm?

> ... our models need to look for these values in the mutable Context, not the const System ...

Since we talking application-layer-god-view already, and we mutable pointers to all of the Systems (and heck, we're going to be adding & removing bodies via calling Plant methods anyway), would your problem be solved if you could call setters on the System? Is there a _need_ to customize these value on a per-Context basis (e.g., for working through many speculative cases in parallel)?  It still might be _easiest_ to add them into Context in order to achieve correctness, but I'm asking what your _use case_ demands in the solution space.

amcastro-tri, 2020-05-07:
Really good question!
I think you nailed it though; correctness. I recently found myself fighting with our C++ APIs attempting to add new setters to mute the "system". That is because I found it difficult to enforce correctness (in terms of invalidating somehow the model). Therefore IMO this is a great application for `Parameters` living in the `Context`. 

E.g. Say the application layer decides to stop the sim to update the mass. If I do `plant.set_mass(&context, body, new_mass)`, I can right away start my sim from where I left off.
If I do `plant.set_mass(body, new_mass)`, nothing gets recomputed until the configuration q gets changed. How bad is that? who know, it'd definitely require more study. It might be more complex yet for other cases.

IMO, it is time for use to make good use of `Parameters`. Event handling IMO is a separate issue though and most users will be able to get away without it (this might not be true for the "adding new floating bodies" request, which is side-to-side with the parameter request according to that meeting).

jwnimmer-tri, 2020-05-07:
As far as correctness goes, a System whose class-member data has changed can trigger an event that ends up invalidating whatever parts of the Context are stale.  It's solveble.  Context may still be an _easier_ solution for correctness, but it's not _hard_ to do it with class-member data.  If you think about it, that _has_ to be true because otherwise how would I/O systems work (e.g., LcmSubscriber) to communicate with non-framework entities?

It sounds like from your use case you don't care either way it's phrased in an API, you just want it to be correct without you worrying about it?

sherm1, 2020-05-07:
Mutable parameters should be (and already are) in the Context, where they are conveniently integrated with the caching system. Modifying the System itself generally requires creating a new compatible Context rather than continuing with the old one.

If we don't care about modifying the Parameters from event handlers, then all we have to do is make more use of Parameters in MBP and we're good. And that does at least maintain a distinction between discrete state and parameters -- parameters are those variables that cannot be updated by any event handler.

We may need to improve the ability of event handlers to interrupt the simulation in a friendly way to permit application-level modification of Parameters, but that is already planned. (And the Simulator's monitor() can already do that.)

Alejandro makes good points above. I would still like to hear from @edrumwri and @naveenoid for their thoughts.

sherm1, 2020-05-12:
Closing out this discussion in favor of narrowing it in #13289.

closed 2020-05-12 by sherm1