RobotLocomotion/drake #13251: Pull in stochastic scenarios

jwnimmer-tri, 2020-05-07:
The Drake developers at TRI have the following capabilities in our internal repositories:
- specify random distributions and transforms in C++, Python, and YAML,
- specify constraints on pose, generalized position, and vector direction in C++, Python, and YAML,
- populate these values into a simulation's initial conditions and diagram-level constraints.

In principle, we are happy to share it, but making it public takes time and we have not yet prioritized it. Please upvote it by adding a "thumbs up" reaction to this comment if you want us to know that you are keen.

Here's an illustration of how it could be used to specify a simulation:
```yaml
# An example scenario for acrobot, with randomness.
one_example:
  controller_params: !UniformVector
    min: [4, 40, 4, 0.9e3]
    max: [6, 60, 6, 1.1e3]
  initial_state: !UniformVector
    min: [1.1, -0.1, -0.1, -0.1]
    max: [1.3, 0.1, 0.1, 0.1]
  t_final: 30.0
  tape_period: 0.05
```

hzyjerry, 2020-05-28:
I am very interested in this for my research. Would love to see progress on it :)
Also happy to chat more and potentially contribute.

jwnimmer-tri, 2020-08-28:
Bumping to high because Russ wants #13282 and this is a prereq.

jwnimmer-tri, 2020-09-02:
I believe everything we need is now available on master, without restriction.  I still some a few clean-ups to follow up, but the feature should be fully operable now.

PR #13991 (2020-09-02): Tidy up nits on schema classes
In #13943 and #13949 there were some clean-ups suggested.  This realizes those requests.

Commits are separate for review, but will be squashed before merging.

Closes #13251.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13991)
closed 2020-09-02 by jwnimmer-tri