RobotLocomotion/drake #13280: mbp: CollectRegisteredGeometries(bodies) should work pre-finalize?

EricCousineau-TRI, 2020-05-12:
I thought that the resolution of #9754 fixed my desired workflow, which is to simultaneously allow adding collisions and collision filters as one is constructing a MBP+SG pair.

However, it looks like `CollectRegisteredGeometries` may still require finalization:
https://github.com/RobotLocomotion/drake/blob/f234c82a6d068fca97125c3cb2804831d82374cb/multibody/plant/multibody_plant.cc#L479-L493

I naively tried to remove this, but it didn't work out immediately (trying to figure out if this is due to downstream refactoring issue, or something else).

\cc @amcastro-tri @SeanCurtis-TRI

SeanCurtis-TRI, 2020-05-12:
As this came across my desk, I had to register surprise. Clearly, registering collision filters works pre-finalize. The [parsing does that](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/detail_urdf_parser.cc#L208-L232).

So, it's not a question of whether collision filter groups can be specified pre-finalize. It's a questions of spelling. What you may not realize is you don't actually need the `GeometryId`s. Just getting the `FrameId` for every body and pushing *those* into the `GeometrySet` is sufficient. Passing a `FrameId` is short-hand convenience of explicitly enumerating the geometries that have been affixed to it.

So, unless you're doing some very *targeted* geometry-based collision filter groups (quite unlikely) just go with the `FrameId`s.

It's worth noting that if you *subsequently* add geometries to that body, they will not be included in the collision filter group.

SeanCurtis-TRI, 2020-05-12:
Perhaps this would be better as a StackOverflow question so others can learn from this?  :)

EricCousineau-TRI, 2020-05-12:
Ooh, great point! Will try that out!

> Perhaps this would be better as a StackOverflow question so others can learn from this? :)

Possibly so! However, this is on GitHub, so it is publicly searchable ;)
But once I make this work out, then yeah, I can post it!

EricCousineau-TRI, 2020-05-12:
FTR I may not have time to try this out in the first iter. Will revisit in a future "refactoring".

EricCousineau-TRI, 2020-05-12:
I lied, I tried out using `geometry_set.Add(*plant.GetBodyFrameIdIfExists(body))` instead of `geometry_set.Add(plant.CollectRegisteredBodies({&body}))`. Both seem to work in an equivalent fashion, which is great!

However, now it seems like if you do collision filtering, you have to do it *after* you have added all the relevant collisions to a body. It kinda makes sense regarding implementation, but t did surprise me a bit. Will create a small separate issue.

PR #13281 (2020-05-13): mbp: Allow CollectRegisteredGeometries to be called pre-finalize
Resolves #13280

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13281)
closed 2020-05-13 by amcastro-tri