RobotLocomotion/drake #13284: Mac Debug builds are running out of disk space

BetsyMcPhail, 2020-05-12:
For example: 

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-mojave-clang-bazel-nightly-debug/301/
https://drake-jenkins.csail.mit.edu/job/mac-catalina-clang-bazel-nightly-everything-debug/209

```
[5:42:41 AM]    ------------------------------------------------------------------------------
[5:42:41 AM]    *** Disk usage for /tmp:
[5:42:41 AM]   Filesystem     Size   Used  Avail Capacity iused      ifree %iused  Mounted on
[5:42:41 AM]  /dev/disk1s1  120Gi  108Gi  9.7Mi   100% 1014684 1255228116    0%   /System/Volumes/Data
[5:42:41 AM]    ------------------------------------------------------------------------------
```

BetsyMcPhail, 2020-05-12:
Note: Disk usage for debug builds was previously high, e.g.:

```
[9:30:35 AM]    ------------------------------------------------------------------------------
[9:30:35 AM]    *** Disk usage for /tmp:
[9:30:35 AM]   Filesystem     Size   Used  Avail Capacity iused               ifree %iused  Mounted on
[9:30:35 AM]  /dev/disk1s1  120Gi  115Gi  3.6Gi    97% 1434875 9223372036853340932    0%   /
[9:30:35 AM]    ------------------------------------------------------------------------------
```

jamiesnape, 2020-05-12:
We recovered about 3G on the new images that will hopefully tide us over while we expand the images, though they cannot get too much larger without reducing the number of nodes we have.

closed 2020-05-15 by jamiesnape