RobotLocomotion/drake #13289: Parameterize physical properties to permit changes to existing models

sherm1, 2020-05-12:
After much discussion (refs below) it is clear that users need to be able to modify basic physical properties in their models without having to rebuild from scratch for each change. It is OK for these changes to be performed at the application level (e.g. between `Simulator::AdvanceTo()` calls) so we do not need to create any new event handlers. Properties with strong user interest:
- mass properties
- contact/friction properties
- joint properties

Some of these will be much easier to modify than others, so we will not attempt all of them at once. This is going to require a series of Drake subprojects. This issue is for identifying those subprojects and making them concrete enough to have their own issues. 

Here is my starting proposal -- please comment.

#### Project 1: Parameterize body mass properties (mass, com, unit inertia)
- This requires only localized modifications to MultibodyPlant (MBP). 
- Need to settle on an API that distinguishes setting/getting "default" mass properties in MBP from setting the runtime Parameter values in the Context. SceneGraph (SG) provides a precedent -- overloaded pairs of methods are provided distinguished by whether the first argument is a Context.
- Modify MBP to declare suitable Parameters to hold the runtime values of the mass properties. Must decide on appropriate granularity, and whether these are scalar-type-convertible numerical parameters or abstract parameters (related issue: #11917).
- MBP cache entries need to declare dependency on the new Parameters.
- Q: should default values also be upgraded to type T or kept as doubles?

#### Project 2: Parameterize per-geometry specification of contact properties
- See issue #13064 for discussion.
- These properties should be maintained by SG as ProximityProperties, not in MBP.
- Modify MBP to obtain stiffness, damping, and friction properties from SG rather than storing them internally.
- Once these are moved, SG already provides the necessary interface for modifying properties in the Context (specifically GeometryState rather than as Parameters). But see issue #9501 -- there is some pressure to get _rid_ of GeometryState!
- Consider dependency issues (cache invalidation). MBPs force/acceleration computations must be invalidated upon a change of contact properties. Currently SG has only a single QueryObject output port; MBP's force computations must depend on that. Will we need more granularity?

#### Project 3: Parameterize MultibodyElement properties (joints, forces, etc.)
- This should be easier than Project 2 but is lower priority for users.
- Especially for System Id, parameterize joint properties like damping and stops and force properties like spring and bushing stiffnesses.
- Consider how to do this in general for new elements -- likely MBElement should provide an API for elements to allocate Context resources.
- Likely this should be split into several subprojects.

Refs: 
[1] Mutating Multibody Models Meeting ([notes](https://docs.google.com/document/d/1a4y_Zgph5gUqPoZrnln_B7qfteFP4TOhPteqr8BpAQk))
[2] Followup discussion at Drake Developers Meeting (see [May 11 2020 entry](https://docs.google.com/document/d/15iNBw_2O3TGbThHr6wlJNKctHSY5bEvt1rLhl5obZao/edit)) 
[3] Discussion in issue #13247 (the current issue replaces that one).

sherm1, 2020-06-10:
From f2f with @amcastro-tri:

There are four "levels" of parameterization to consider for MultibodyPlant:
1. Default values stored in the MBPlant object (always),
2. copies of those defaults in the MBTree object (post-finalize),
3. default-constructed Contexts, and
4. Contexts containing Parameter values.

Ideally every default value in an MBPlant has a corresponding Parameter resource in the Context. The value of that Parameter is set to the MBPlant default when first allocated, but can be changed independently in that Context subsequently. All computations in MBPlant are done using the Parameter values, _not_ the default values. In that case modifying default values in an MBP System object does not invalidate any existing Contexts for that System. If the MBP system has already been finalized the new default value should propagate to the internal MBTree object. New Contexts allocated after the default value modification will have their Parameters initialized to the new defaults while old Contexts simply retain their old Parameter values. No existing computation is invalidated by the change.

The situation is much trickier if there is a "default" value in MBPlant but no corresponding Parameter in the associated Contexts. Imagine, for example, that we make the change proposed in #13520 allowing WeldJoint X_PC to be changed in the MBPlant but don't also add an X_PC Parameter to the Context. In that case MBPlant performs computations (e.g. kinematics) using its internal X_PC value; the results of those computations are cached in the Context. A subsequent change to the internal X_PC _should_ invalidate the cached computations since they are now clearly wrong. However, there is no direct connection between System objects and Context objects so that invalidation cannot be performed and we now have Contexts around whose cache entries think they are up to date but of course they are not. Needless to say that would be bad!

IMO the best way to deal with this is to make sure any settable values in MBP have corresponding Parameters in MBP's Contexts. In cases where that isn't possible, the only available solution is to modify the MBP System's stored id number. When a system API is given a context, it checks that the system id stored in the context is the same as the stored id and throws if not. That is a brute force way to make sure no out-of-date computations are used. We can sweeten this a little by providing an API to update the stored system id in an existing Context, which API would have the side effect of invalidating all cache entries in that Context.

An alternative to consider would be that we _always_ change the system id after any call to a non-const System API, and provide `system.ResurrectContext(&context)` or something like that to make necessary repairs. With some effort, we could make it so that method knows whether cache entries must be invalidated or not.

@joemasterjohn: some things to worry about!
fyi @rpoyner-tri

jwnimmer-tri, 2020-06-10:
Another instance of the rule: be suspicious if your System has non-const member fields, especially if they are referred to by Calc methods.

EricCousineau-TRI, 2020-09-15:
I forgot - was changing frame offsets part of this too?

Asking b/c I was wondering about changing the "setpoint" (grounded pose of `frameA`) of `LinearBushingRollPitchYaw` as enabled by #14000

joemasterjohn, 2020-09-15:
Yes that's the next thing I'm currently working on (changing `X_PF` for a `FixedOffsetFrame`). I'll let you know when I have a PR up for that.

EricCousineau-TRI, 2020-09-15:
Thanks! I was able to workaround it for now by checking when I set the diagram state, but it'd be awesome to use it as a parameter!

EricCousineau-TRI, 2023-04-15:
Any chance we can queue this back up, or at least a portion of it?
Specifically, this point, esp. in relation to #18932:

> Especially for System Id, parameterize joint properties like damping and stops and force properties like spring and bushing stiffnesses.

In this case, it's very useful to be able to vary the joint damping.
`set_default_damping` is a bit awkward as that has to be called pre-Finalize, and is not consistent with things like sys ID on inertial parameters.

jwnimmer-tri, 2024-11-01:
@sherm1 @joemasterjohn I think it would be worth posting an update here summarizing which parameters _can_ already be changed (both in the tree and in the context), and listing out the remaining parameters that are still missing setter functions.

I think probably we should explode this issue and file smaller issues for each missing individual thing (and we have some duplicates of those already) but let's start by writing down the list here.