RobotLocomotion/drake #13296: systems lcm: Calling DrakeLcm::Publish() before Simulator::Initialize() causes initial events to be lost?

EricCousineau-TRI, 2020-05-13:
I was tinkering with Anzu stuff (Python sim) and adding debug print messages. However, I found that I was sometimes losing my print messages when calling `Simulator.Initialize()`.

@jwnimmer-tri or @sherm1 Do y'all have any ideas on this?

## Basic elements

- `LcmInterfaceSystem` and `LcmSubscriberSystem` are linked (same `LCM` instance) and added to a diagram
- Publish is called before `Initialize`

## Workaround

Do not publish until after `Initialize`

## Min Repro / MWE

Here is the minimum reproduction, that was based on commit 82057e6789:

```py
from robotlocomotion import quaternion_t

from pydrake.lcm import DrakeLcm
from pydrake.systems.analysis import Simulator
from pydrake.systems.framework import DiagramBuilder, LeafSystem
from pydrake.systems.lcm import LcmInterfaceSystem, LcmSubscriberSystem

class PrintTime(LeafSystem):
    def DoPublish(self, context, events):
        print(f"t: {context.get_time()}")

def main():
    lcm = DrakeLcm("memq://")
    lcm_type = quaternion_t
    channel = "JUNK"
    dt_sec = 1.

    builder = DiagramBuilder()
    builder.AddSystem(PrintTime()).DeclarePeriodicPublish(dt_sec)
    lcm_system = builder.AddSystem(LcmInterfaceSystem(lcm))
    builder.AddSystem(LcmSubscriberSystem.Make(channel, lcm_type, lcm))
    diagram = builder.Build()

    for do_publish in (False, True):
        print(f"--- do_publish: {do_publish} ---")
        simulator = Simulator(diagram)
        if do_publish:
            lcm.Publish(channel, lcm_type().encode())
        print("Initialize")
        simulator.Initialize()
        print("AdvanceTo")
        simulator.AdvanceTo(dt_sec)
        print()

if __name__ == "__main__":
    main()

"""
--- do_publish: False ---
Initialize
t: 0.0
AdvanceTo
t: 1.0

--- do_publish: True ---
Initialize
AdvanceTo
t: 1.0

"""

```

jwnimmer-tri, 2020-05-13:
The call to `DrakeLcm::Publish` will cause `LcmInterfaceSystem::DoCalcNextUpdateTime` to return `context.get_time()` instead of `infinity`.  Perhaps that interacts with how the simulator processes initialization publication events.  We have two events both occurring at `time == 0`, maybe it is only executing one of them before moving on.

EricCousineau-TRI, 2020-05-13:
Ah, gotcha. So it updates `DoCalcNextUpdateTime()`, and then the initial `PrintTime.DoPublish` implementation gets skipped (since it's no longer part of the events triggered by `Simulator.Initialize`)?

jwnimmer-tri, 2020-05-13:
I think its because of the time warp.  The `LcmInterfaceSystem::DoCalcNextUpdateTime` would be reporting `-denorm_zero`, but the periodic event is at exactly zero.  The Diagram throws away the second event because its strictly greater, but the simulator assumes that denorm zero and zero and equalivalent and never asks about zero again.

Time warp:
https://github.com/RobotLocomotion/drake/blob/462b7ec8de4a5bfa3aa11248ca937dbe2a3e2e72/systems/analysis/simulator.cc#L98-L102

jwnimmer-tri, 2020-05-13:
I am not sure whose fault this is.  I'm tempted to say that the Diagram needs to treat denormally-separated event times as equivalent (i.e., rounding them) and not discard the exactly-zero events during merge. Or the integrator needs to advise the diagram about a tolerancing for what is equivalent or not. Probably requires more careful study, though.

jwnimmer-tri, 2020-05-13:
In a single-threaded, memq-only example (no network races) such as the snippet above, calling `LcmInterfaceSystem::HandleSubscriptions(0)` would render the `LcmInterfaceSystem` inert as the simulation begins, even if traffic had already been sent.  That would be another work-around, but would be risky in any situation with the URL was memq, or where there was an async tx thread writing to the memq.

sherm1, 2020-05-13:
The time warping in Simulator::Initialize() carefully _avoids_ denormalized numbers. It is setting time to the first normalized number less than the start time, in the (unwritten!) expectation that an overridden `DoCalcNextUpdateTime()` would never return a time before its start time. [`LcmInterfaceSystem::DoCalcNextUpdateTime()`](https://github.com/RobotLocomotion/drake/blob/master/systems/lcm/lcm_interface_system.cc#L57-L73) violates that assumption and asks for a time prior to the start time (by reading it out of the Context). 
```c++
*time = context.get_time();
```
A quick (not elegant) fix would be to change that to
```c++
*time = std::max(0., context.get_time());
```

@edrumwri: check it out -- an unexpected consequence of the previous-normalized-number trick in Simulator::Initialize()! Any thoughts?

sherm1, 2020-05-13:
Oh, wait -- I think I see a much nicer fix. In `Simulator::Initialize()` change:
```c++
if (next_timed_event_time_ == current_time) {
  time_or_witness_triggered_ = kTimeTriggered;
}
```
to
```c++
if (slightly_before_current_time <= next_timed_event_time_  
    && next_timed_event_time_ <= current_time)
  time_or_witness_triggered_ = kTimeTriggered;
```
That would allow a DoCalcNextUpdateTime() that return `context.get_time()` to work unchanged here.

What do you think Evan?

jwnimmer-tri, 2020-05-13:
In my mental model, neither of those fixes works.  I think the crux of the problem is that Diagram is not even returning the time==0 event back to the Simulator.  It's probably worth making a min repro in C++ (e.g., in a simulator_test unit test) and iterating from there.

EricCousineau-TRI, 2020-05-13:
>  It's probably worth making a min repro in C++ [...]

Given that we know the root cause, this is probably no longer minimal, but I posted a C++ versino of the above program here:
https://github.com/EricCousineau-TRI/drake/blob/45ea702d5e6b1a25facc1b07c8809166c19e4b36/tmp/repro.cc

sherm1, 2020-05-13:
I ran the C++ version and hacked on it a little. Using `max(0, context.get_time())` (in both LcmInterfaceSystem and LcmSubscriberSystem) does fix the problem. My "better" fix doesn't work.

Here is what's happening at initialization:
- There are 3 systems with CalcNextUpdateTime() contributions.
- The Printer system's first update time is always 0.
- The Lcm systems return t=∞ if there is no pending message, t=current_time otherwise.

Simulator::Initialize() sets current_time to -1e-308 prior to calling Diagram::CalcNextUpdateTime(). That method interrogates the three subsystems and notes the earliest next publish time. When there is no pending message the next time is t=min(0,∞,∞) so only the Printer system has a pending event then. When there is a pending message, the next time is t=min(0,-1e-308,-1e-308) so only the Lcm systems are involved -- Printer has no publish at the earlier time.

Then Initialize() resets time to the actual start time of 0 and resumes. AdvanceTo() is called. Printer's next update time _after_ 0 is 1 so we never see the t=0 publish.

I'm not sure what a good fix is here. Ignoring claims of events occurring before time zero would fix the problem, but @RussTedrake recently allowed for times to be negative (though still moving forward) #13061. Presumably we want events to continue working with negative times.

Does this jog any ideas loose, @edrumwri ?

sherm1, 2020-05-26:
Any fix for this problem needs to address these needs:
- During `Initialize()`, but not `AdvanceTo()`, scheduled events whose schedule includes _current_time_ should trigger.
- Under all circumstances, unscheduled events that must occur _right now_ should trigger (that's what the Lcm system overrides do when a message arrives).
- The _right now_ events should not preclude scheduled events at the same time (the reported bug).

Possible fixes:
1. When collecting the next events reported by CalcNextUpdateTime(), treat events at both current time t and t+δ as simultaneous and occurring at t+δ. Here t+δ is the next representable floating point number greater than t. Typically we'll have δ=(1+ε)t where ε~2.22e-16 is machine precision, although δ is much smaller when t is near 0 as it was in the presenting case above. The (insignificant?) downside is that this does not distinguish Initialize() from normal step use of CalcNextUpdateTime() so means we will always treat event times t and t+δ as simultaneous at t+δ. There is also some squirrelly code in Diagram::DoCalcNextUpdateTime() for fiddling with the event times.

2. Add a field to the Context time struct that holds the original time T when Initialize() has perturbed it to t̅=T−δ. Then any event that claims to be at time t̅ we interpret as though it had specified time T instead. During ordinary stepping T and t̅ are the same time so this has the advantage of only affecting Initialize(). The downside is the obscure extra field in Context.

3. Add a flag to CalcNextUpdateTime() that says whether we're initializing or not. This would be passed through to DoCalcNextUpdateTime() overrides that can modify their behavior accordingly. The downsides would be (a) the change in signature of a virtual method, generally meaning a flag day to fix, (b) modification to LeafSystem's periodic event scheduling code in its default DoCalcNextUpdateTime(), and (c) changes to overridden DoCalcNextUpdateTime() methods if they attempt to schedule periodic or timed events.

I am inclined to implement fix (2) above as least weird and having no mysterious effects on simultaneity. However, if anyone is in favor of hack (1) in the interest of not mucking with the Context I'd consider that. Fix (3) is the way Simbody does it, but it started out that way. I think it is too extreme a retrofit for Drake.

Thoughts?
@edrumwri @EricCousineau-TRI @jwnimmer-tri

PR #13438 (2020-06-02): Make CalcNextUpdateTime() "right now" returns work during initialization
Implements solution (2) to the event timing problem reported and discussed in #13296.

- Adds an std::optional `true_time` field to Context, and internal PerturbTime() method for setting it.
- Modifies Simulator::Initialize() to use PerturbTime().
- Modifies System::CalcNextUpdateTime() to report true_time rather than the perturbed time.
- Adds unit tests that duplicate the reported problem.
- Cleans up some time-related Context code.

Resolves #13296

---
This change is [[image]](https://reviewable.io/reviews/robotlocomo [... cut]
closed 2020-06-02 by sammy-tri