RobotLocomotion/drake #13299: Should we include MixedIntegerBranchAndBound in ChooseBestSolver ?

RussTedrake, 2020-05-13:
I would vote "yes" because we do not support any MIP solvers in our binary releases.  It would be nicer to write code that just formulates a MIP, and dispatches to gurobi, etc, if available but falls back to our own branch and bound.

@hongkai-dai , @TobiaMarcucci -- what do you think>?

RussTedrake, 2020-05-13:
If we think yes, then conceptually it would be nice to use `ChooseBestSolver` again to pick the solver_id that the branch and bound uses.  If `ProgramAttributesSatisfied` accepted a `ProgramAttributes` instead of a `MathematicalProgram`, then this would be easy... we would just remove the binary variables attribute and call it again.

TobiaMarcucci, 2020-05-13:
I'd also vote yes. Students were a little confused by the way that Hongkai's B&B is currently called, they thought it was a "different way to solve an optimization problem," and not another solver. This made some confusion when I was telling them that Gurobi actually uses a B&B algorithm.

hongkai-dai, 2020-05-13:
I will add it to ChooseBestSolver, probably after Jeremy's PR #13279 lands.

jwnimmer-tri, 2020-05-13:
As far as how to accomplish this, I think the plan should be to add a `class FooSolver : public SolverBase` that calls `ChooseBestSolver` (on the simplified program) then `MixedIntegerBranchAndBound` during `FooSolver::Solve`.  Then `ChooseBestSolver` can know about `FooId` (presumably at the end of the list).

RussTedrake, 2020-05-13:
Is there a reason that `MixedIntegerBranchAndBound` does not derive from `SolverBase`?

jwnimmer-tri, 2020-05-13:
As it stands now, `MixedIntegerBranchAndBound` has-a `MathematicalProgram` and provides for the same kind of data as `MathematicalProgramResult`.  While we could refactor it to derive from `SolverBase`, a wrapper seems easiest and in line with how we treat other servers -- we have `FooSolver` for the wrapper and the server is a distinct library unit tested on its own.

jwnimmer-tri, 2021-11-11:
Update as of #15541: when try to solve a MIP but no solver can do it (e.g., Guribi is missing), now at least the failure message tells users that they can try `MixedIntegerBranchAndBound` by hand.  This is not necessarily as smooth as we desired upthread, but possibly gets some users unstuck on their own.

closed 2024-11-01 by jwnimmer-tri