RobotLocomotion/drake #13309: workspace: hard code paths instead of using pkg-config

jamiesnape, 2020-05-14:


EricCousineau-TRI, 2020-05-27:
If possible, I would like to request that hard-coded paths be hoisted to top-level constants, easy to discover if possible, and easy to hack with (with standard disclaimer of at-your-own-risk).

Example workflow:
https://github.com/RobotLocomotion/drake/pull/13027#issuecomment-611809521

jwnimmer-tri, 2020-05-27:
Would a `local_repository_override` parameter meet the need?  That would allow for a certain uniformity across different rule types.

Example for github archives:
https://github.com/RobotLocomotion/drake/blob/6a068e39a6e4893b53058caf0a812403dd6c3831/tools/workspace/github.bzl#L40-L43

EricCousineau-TRI, 2020-05-27:
I think so, as long as it can affect all occurrences of the Python interpreter; I'm guessing you could do that by generating a Skylark file with that macro, and then pull it back out?

jwnimmer-tri, 2020-05-28:
Ah, I see now.  I think you are talking about python interpreter paths, not pkg-config paths.  This issue is just about pkg-config.

However, I think you're right in any case that if more info about the python interpreter ends up being hard-coded, we should be sure there's a mechanism to adjust it for experimentation.  But I think as long as we follow DRY like we usually do, the places to adjust for experimentation will be few.

EricCousineau-TRI, 2020-05-28:
Sounds good. Sorry to clutter the scope, just hopped through the philosophical link from #13400.
Thanks!

jwnimmer-tri, 2023-01-18:
After #17231 wraps up, we will be using much less via `pkg-config`.  What remains will be approximately this:

```
eigen
fmt
glew
glib
blas
lapack
spdlog
x11
zlib
```

I think those will be sufficiently safe to leave alone without much risk.

closed 2023-01-18 by jwnimmer-tri