RobotLocomotion/drake #13313: Add system scalar conversion with new type as a template argument

jwnimmer-tri, 2020-05-14:
In #13131 we discussed a missing feature from `System<U>` -- the ability to scalar convert it into a `System<T>` where `<T>` is a template argument, instead of being part of the method name as currently such as `ToAutoDiffXd`.

> Adding a templated To... converter is fine. Ideally, we'd offer all three flavors (static, maybe, throws) and use analogous spellings to the current `AutoDiffXd` and `Symbolic` methods, for consistency. (Assuming the static one's more-templated still makes sense; I haven't penciled it out.)

This is an issue to request the new API.

See https://github.com/RobotLocomotion/drake/issues/13131#issuecomment-628578100 for a little more background.

EricCousineau-TRI, 2020-05-14:
Peanut gallery comment:
Here's a (more constrained) analog in Python:
https://github.com/RobotLocomotion/drake/blob/4ca6218ebff1aed9bf096ea01589899e9804015c/bindings/pydrake/multibody/test/plant_test.py#L112-L124

I'm guessing C++ would work the same way - just query the virtual override using `typeid(T)` at runtime?

jwnimmer-tri, 2020-05-14:
In C++ I would expect `if constexpr (std::is_same<T, AutoDiffXd>)` to be the trick, not `typeid`.

hongkai-dai, 2020-06-23:
If this feature works, then we can easily get both AutoDiff with constant size, or the Hessian using nested autodiff.

jwnimmer-tri, 2020-06-23:
This would still be a closed set of support scalar types.  This is just allowing a method call syntax where the one (of three) choices is a template argument, instead of part of the method name (i.e., the ToAutoDiffXd, ToSymbolic, etc. family).

As always, if we want a new scalar in Drake, then we add it to `default_scalars.h` and throughout the code.

rpoyner-tri, 2021-04-30:
I've begun a sketch of this; the C++ is pretty straightforward. Question:
what conversion pairs must be supported? Obviously:

double --> autodiff
double --> symbolic

Any others that should be supported? self (e.g., double --> double) transitions? * --> double transitions?

jwnimmer-tri, 2021-04-30:
I think the starting point would be to match the conversions that are available by default in SystemScalarConverter:
https://github.com/RobotLocomotion/drake/blob/v0.29.0/systems/framework/system_scalar_converter.h#L102-L110

rpoyner-tri, 2021-04-30:
Excellent! The C++ is mostly done. No 'if constexpr's were harmed in the production of this patch.

Onward to pydrake land.

PR #14986 (2021-05-03): system: Support scalar conversion by template parameter
Closes: #13313.

Adds various ToScalarType<U> methods to System, after the manner of
To{AutoDiffXd,Symbolic}, and refactors all of them to use the single templated
implementation. Also adds python bindings and unit tests.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14986)
commit 6fd4443b (2021-05-03): system: Support scalar conversion by template parameter (#14986)

Closes: #13313.

Adds various ToScalarType<U> methods to System, after the manner of
To{AutoDiffXd,Symbolic}, and refactors all of them to use the single templated
implementation. Also adds python bindings and unit tests.
closed 2021-05-03 by jwnimmer-tri