RobotLocomotion/drake #13316: Port examples/rod2d to SceneGraph

jwnimmer-tri, 2020-05-14:
Currently, `examples/rod2d` uses the attic-based drake visualizer glue.  We should port it to use `SceneGraph` instead.

Relates #12158.

https://github.com/RobotLocomotion/drake/blob/4ca6218ebff1aed9bf096ea01589899e9804015c/examples/rod2d/rod2d_sim.cc#L88-L91

jwnimmer-tri, 2020-05-14:
@SeanCurtis-TRI To confirm my understanding, is the best thing to do here to write a new `RodGeometry` utility class, akin to [PendulumGeometry](https://github.com/RobotLocomotion/drake/blob/master/examples/pendulum/pendulum_geometry.h)?

(I'm hoping we can find an interested new developer to work on this as an onramp issue, not you or I.)

SeanCurtis-TRI, 2020-05-14:
Yes and yes. We happen to be "on boarding" someone at the beginning of June.  :)  @sherm1 are you thinking what I'm thinking?

jwnimmer-tri, 2020-05-14:
I was going to ask @IanTheEngineer if he wants to look next week.  Might be a good onramp for RViz work.

SeanCurtis-TRI, 2020-05-14:
I wouldn't say no to that.

sherm1, 2020-05-15:
Sounds like a good mini-project. I'm not sure whether Rod2d is actually a great intro to modern Drake, but just making it talk to SG would provide some useful Drake know-how.

SeanCurtis-TRI, 2020-05-15:
The nice thing is that there are numerous examples of the `PendulumGeometry` pattern in Drake. So, without delving into the details of the system implementation too deeply, you can visualize.

jwnimmer-tri, 2020-08-17:
We need this finished in the next two weeks, or else we'll remove rod2d from master on 2020-09-01.

SeanCurtis-TRI, 2020-08-17:
I'll see if I can't just fix it this afternoon. Probably faster to change than to discuss.

PR #13890 (2020-08-19): [examples] Connect rod 2d to SceneGraph for visualization
Supplant the old visualization mechanism with the standard used in examples. examples/rod2d is no longer dependent on attic code.

resolves #13316

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13890)
closed 2020-08-19 by SeanCurtis-TRI