RobotLocomotion/drake #13320: bazel_external: `drake` is on the Python path, and `common/__init__.py` is auto-created

EricCousineau-TRI, 2020-05-14:
This is the more specific version of this comment: https://github.com/RobotLocomotion/drake/issues/7871#issuecomment-628910452

From this @pvarin's StackOverflow post:
https://stackoverflow.com/q/61802860/7829525

To reproduce, see my updated example:
[`drake_bazel_external/apps/bar.py`](https://github.com/EricCousineau-TRI/drake-external-examples/blob/1165e06b243aa095996c3d019d84d036d6515812/drake_bazel_external/apps/bar.py#L27-L64) ([branch](https://github.com/EricCousineau-TRI/drake-external-examples/blob/feature-pv-name-conflict))

When running, this is the output that I see:
```sh
$ bazel build //apps:bar
$ alias bash-isolate='env -i HOME=$HOME DISPLAY=$DISPLAY SHELL=$SHELL TERM=$TERM USER=$USER PATH=/usr/local/bin:/usr/bin:/bin bash --norc'
$ bash-isolate -c bazel-bin/apps/bar

path:
  {source_tree}/apps
  {runfiles}
  {runfiles}/drake/bindings
  {runfiles}/lcmtypes_bot2_core/lcmtypes
  {runfiles}/lcmtypes_bot2_core
  {runfiles}/lcmtypes_robotlocomotion/lcmtypes
  {runfiles}/lcmtypes_robotlocomotion
  {runfiles}/meshcat_python/src
  {runfiles}/spdlog
  {runfiles}/meshcat_python
  {runfiles}/lcm
  {runfiles}/ignition_math
  {runfiles}/drake
  {runfiles}/drake_external_examples
  /usr/lib/python36.zip
  /usr/lib/python3.6
  /usr/lib/python3.6/lib-dynload
  /usr/lib/python3/dist-packages

common: {runfiles}/drake/common/__init__.py

Traceback (most recent call last):
  File "{runfiles}/drake_external_examples/apps/bar.py", line 24, in <module>
    from common import foo
ImportError: cannot import name 'foo'
```

Note that `common` should point to `drake_external_examples/common/__init__.py`, not `drake/common/__init__.py`.

FTR The overall correct decision is to import using the workspace name. Changing:
```py
from common import foo
```
to
```py
from drake_external_examples.common import foo
```

FYI @jwnimmer-tri

jwnimmer-tri, 2020-05-14:
With the "Issue 1, 2" callouts above, I can understand what's happening any why it causes friction.

I think `legacy_create_init` mentions before were confusing me.  I don't see how that's relevant here.  For example, it would be easy to depend on something in drake that brought in even a real `__init__.py` that we wrote for ourselves.  In general `from drake.stuff import other_stuff` should work.

It seems to me the only problem is Issue 1 biting us.  To mitigate, we either need to (a) change how bazel python sets up the import path project-wide, or (b) show users how to correctly spell imports in the bazel world.

Since (a) seems like a project wide, rcfile-type setting, it's not fixable in Drake anyway.  Maybe a prototypical bazelrc in drake-external-examples.

However, doing (b) should be easy and sufficient to resolve the question.  I think the only action here is to add a `py_library` to `drake-external-examples` that shows how to import your own python code and pydrake code at the same time?

EricCousineau-TRI, 2020-05-14:
>  I think the only action here is to add a py_library to drake-external-examples that shows how to import your own python code and pydrake code at the same time?

SGTM. Will submit strawman example.

EricCousineau-TRI, 2020-05-14:
Also, for the record, I have one, but it's reeks of "unofficialness":
https://github.com/EricCousineau-TRI/repro/blob/39f79009a2e89b987f072276d1921a282f63e6a1/python/bazel_py_example/mid/py3_bin.py#L3

jwnimmer-tri, 2020-05-28:
Eric is working on the PR for this, so changed assignment.

jwnimmer-tri, 2022-05-06:
The PR https://github.com/RobotLocomotion/drake-external-examples/pull/175 died on the vine.  Closing this as "won't fix".  Any anyone is interested, they can work on resurrecting that PR.

closed 2022-05-06 by jwnimmer-tri