RobotLocomotion/drake #13321: Add oriented bounding box trees for hydroelastic meshes

sherm1, 2020-05-15:
(Per discussions with @DamrongGuoy and @SeanCurtis-TRI; please comment/edit)

### Background
We currently use axis aligned bounding box (AABB) trees as our acceleration data structure for the tri/tet mesh intersection operation necessary for producing the hydroelastic contact patch. This produces a lot of false-positive intersections (measured at 77% in one test), likely because the AABBs are not a good fit for arbitrarily-oriented triangles. The resulting box intersections were measured to consume almost half the CPU time needed to create the contact surface.

### Proposal
Build oriented bounding box (OBB) trees instead. Those should provide a much tighter fit, especially for the tri mesh. Determine the best granularity for the leaf nodes -- likely putting a single triangle in a box is suboptimal.

### Project outline
- Implement a method that makes a tight OBB around a set of points.
- Implement a method to build a tree of OBBs with a parameter controlling granularity for the leaf nodes.
- Research the best-performing leaf granularity.
- Figure out a way to measure and visualize the existing AABB tree and new OBB trees (during development)  so that we can confirm for ourselves and convey to others how much we've improved. 

### References/resources
Simbody's OBB-maker [declaration](https://github.com/simbody/simbody/blob/master/SimTKmath/Geometry/include/simmath/internal/Geo_Point.h#L273)  /   [code](https://github.com/simbody/simbody/blob/master/SimTKmath/Geometry/src/Geo_Point.cpp#L475)

Method there is: centroid and principal moments to get an initial OBB guess, followed by an optimization with orientation as design variables to minimize volume subject to containing all the points.

(The various helper methods are in those same files. Also it is easy to pull Simbody into CLion to facilitate code navigation.)

sherm1, 2020-05-15:
BTW @hongkai-dai probably knows how to compute the _optimal_ bounding box containing a set of points.

hongkai-dai, 2020-05-15:
It doesn't seem that the problem is convex to me. Here is a formulation of the problem
```
min b(1) * b(2) * b(3)
s.t -b ≤ Rᵀpᵢ + d ≤ b ∀ pᵢ
    R ∈ SO(3)
```
where pᵢ is the location of the i'th point. The decision variables are `R, d, b`. The problem here is that SO(3) constraint is non-convex. 

We could either solve this problem through nonlinear nonconvex optimization (hence find a localy optimal box), or I can use a convex relaxation of SO(3), and hence give a good guess on the optimal bounding box through convex relaxation. But I don't think I can guarantee to find the global optimal oriented bounding box.

avalenzu, 2020-05-15:
@hongkai-dai's completely correct. Once you utter the phrase "orientation as decision variables," all hope of convexity goes out the window.

hongkai-dai, 2020-05-15:
We could find the smallest oriented ellipsoid that covers all the points (this is a convex optimization problem), and then use the orientation of the ellipsoid as the initial guess of the oriented bounding box.

jwnimmer-tri, 2020-05-15:
I may be misunderstanding the plan here, but please think twice before adding a dependency from SceneGraph or MultibodyPlant onto MathematicalProgram.

avalenzu, 2020-05-15:
Re. @hongkai-dai: Or even just use the box defined by that ellipsoid's half-axes. There's *a lot* to be said for not introducing NLP's into your pipeline. They're very sensitive (as @hongkai-dai is well aware :smile:).

Re. @jwnimmer-tri: Good point.

sherm1, 2020-05-15:
The code we have lying around is self contained with an amateur optimizer hack, and in practice that may be fine. But if it turned out we needed a better solution to a problem for which MathematicalProgram was well suited, why wouldn't that be the obvious choice rather than attempting to reinvent that wheel?

jwnimmer-tri, 2020-05-15:
For all the same reasons that extra very heavy dependencies and dependency cycles should cause one to pause and reflect on the tradeoffs.

sherm1, 2020-05-15:
Thanks @hongkai-dai and @avalenzu! The minimum containing ellipsoid is an interesting alternative to principal moments for finding an initial orientation, and the ellipsoid would be insensitive to point clusters. Do you have a sense of how expensive that optimization would be? Principal moments is O(n), but to do it nicely for clumped points (might not be necessary for us here) it would require a convex hull calculation first.

hongkai-dai, 2020-05-15:
Finding the smallest ellipsoid would require more than O(n). We will solve a semidefinite programming problem, which is much harder than doing one PCA. What's the magnitude of the number of points are we talking about? If it is around 10 points then I would imagine solving the optimization problem takes a few miliseconds.

sherm1, 2020-05-15:
Per f2f the largest box at the OBB tree root has to surround the whole mesh, which can easily have 10⁵ or 10⁶ points. A pre-convex hull might reduce that substantially for O(n log n), but there would still be a lot of points. OTOH down at the leaves the boxes only need to cover a few tris or tets, so those would have ~10 points.

DamrongGuoy, 2020-07-24:
I update that I have prototyped these two capabilitlies (no code in master yet):

1. Write AABB tree and OBB tree to VTK files for visualization in Paraview. We can confirm for ourselves and convey to others how much we've improved. The left column show AABB tree at depth=4 and at the leaves.  The right column show OBB tree at depth=4 and at the leaves.
[image]
[image]

2. Using the method in Simbody as reference, the prototype calculated an initial OBB using a covariance matrix followed by an optimization with orientation as decision variables to minimize volume subject to containing all the points.

Now I am working on the third capability:

3. Build an OBB tree with a parameter controlling granularity for the leaf nodes.

DamrongGuoy, 2020-08-04:
The prototype is good enough to get into master now. The prototype is in the branch:
[https://github.com/DamrongGuoy/drake/tree/OBB_draft3](https://github.com/DamrongGuoy/drake/tree/OBB_draft3)

Thanks to @SeanCurtis-TRI, we have a strategy to put it into master and minimize Reviewable effort. A proof of cencept is in [commit 847fbfa](https://github.com/SeanCurtis-TRI/drake/commit/847fbfa50948ab6982609146e55417a1c31e4284) and [commit bb9ccdf](https://github.com/SeanCurtis-TRI/drake/commit/bb9ccdf2868daf531147db6042d3796932758a08) of Sean's [branch bvh_curate_proof_of_concept](https://github.com/SeanCurtis-TRI/drake/tree/bvh_curate_proof_of_concept).

The main observation is that the structures of Aabb-related code and Obb-related code are very similar. We will substitute Obb for Aabb to keep only one code of Bvh and rely on Git to remember Aabb, so we can bring it back in the future.

This is the plan for the PR train:

### First PR. Add `class Obb` into master.
Commit1. Copy Aabb to Obb as starting point.
The code of `class Aabb` is in `bounding_volume_hierarchy*.*`, so we'll copy it to `obb*.*` as the starting point. Tell the reviewer that this commit is a straightforward file copying.
```
cp bounding_volume_hierarchy.h obb.h
cp bounding_volume_hierarchy.cc obb.cc
cp bounding_volume_hierarchy_test.cc obb_test.cc
git add obb.h obb.cc test/obb_test.cc
```
Commit2. Add content of Obb.
Manually replace `class Aabb` in obb.h, obb.cc, and obb_test.cc with `class Obb` from the prototype branch [OBB_draft3](https://github.com/DamrongGuoy/drake/tree/OBB_draft3) and edit BUILD.bazel  to compile and run unit tests. The change should have the skeleton of Aabb but with details of Obb in order to minimize Reviewable diff.

### Second PR. Add the new Bvh in master.
Commit1. Rename files `bounding_volume_hierarchy*.*` to `bvh*.*` as starting point.
```
git mv bounding_volume_hierarchy.h bvh.h
git mv bounding_volume_hierarchy.cc bvh.cc
git mv bounding_volume_hierarchy_test.cc bvh_test.cc
```
Commit2. Change `class BoundingVolumeHierarchy` to `class Bvh` and use `Obb` instead of `Aabb`.

2.1. Manually copy content from `bvh*.*` in prototype branch [OBB_draft3](https://github.com/DamrongGuoy/drake/tree/OBB_draft3) to `bvh*.*` to this production branch.
2.2 To minimize Reviewable diff, do these (more details in Sean's proof of concept [commit bb9ccdf](https://github.com/SeanCurtis-TRI/drake/commit/bb9ccdf2868daf531147db6042d3796932758a08)):
- rename various entities,
- restore documentation, and
- add Bvh::Collide with Primitive shape types.

### Visualization in a separate branch for future PR.
In a separate branch, we will keep `bvh_to_vtk.*`(write Bvh to VTK files for visualization in Paraview). It is not a part of this PR train.

SeanCurtis-TRI, 2020-08-04:
Minor clarification on the first PR.

Don't put the *full* contents of `bounding_volume_hiearchy.*` into the `obb.*` files. Just the portions that belong to the `Aabb` class.

DamrongGuoy, 2020-08-04:
Oops!  I already copied the full contents. To fix that, I added another commit to keep only the portions that belong to the `Aabb` class.

DamrongGuoy, 2020-09-09:
Two PRs are in flight: [13808](https://github.com/RobotLocomotion/drake/pull/13808) for Obb and [13978](https://github.com/RobotLocomotion/drake/pull/13978) for Bvh using Obb.

The unit tests show that my code does not always find the optimal oriented bounding box, which is consistent with what @hongkai-dai and @avalenzu mentioned before. I'll file another issue to improve the algorithm in the future.

DamrongGuoy, 2020-09-19:
PR #13808 has merged. PR #13978 should merge soon.

I created two issues to work in the future:

- issue #14067 [geomtry] Improve CalcOrientationByPca() for covariance matrix with repeated eigenvalues.
- issue #14081 [geometry] Improve ObbMaker::OptimizeObbVolume().

The code for visualizing Bvh and Obb is in my own branch (DamrongGuoy/drake/BvhToVtk). There's no plan to merge it into master yet.

DamrongGuoy, 2020-09-22:
PR #13978 was merged. Close this issue now.

closed 2020-09-22 by DamrongGuoy