RobotLocomotion/drake #13324: scene_graph: Should have ability to check if any contexts have been allocated?

EricCousineau-TRI, 2020-05-15:
Related to Anzu PR review with @calderpg-tri.

Per the docs (as of time of writing): https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_scene_graph.html

Given that you can modify a model value, but it may not affect previously affected contexts, it would be nice to know if code modifying a model value may not percolate to *all* contexts (example application: you're adding new collision geometry, etc.).
Currently, the "simplest" way to do this is to check if the associated `MultibodyPlant` is finalized; that's rather indirect / imprecise.
Ideally, there's a concise way to interrogate a `SceneGraph` if it's allocated any contexts.

Possible solutions:
* (Acute, const-cast-dirty) `SceneGraph` gains an internal (mutable) boolean that's altered by `AllocateContext`.
* (Acute, but cleaner-ish) `SceneGraph` gets a (local-to-that-instance) counter for "how many internal::GeometryStates do I have", and `AllocateContext` will ultimately tick that counter. If this count is more than one, then 
* (General) `System` gains the "cleaner-ish" approach, for everything in general.

@SeanCurtis-TRI @sherm1 Do y'all have any prefs?
I presented it for "general" Systems just in case of the MBP/Systems mutation stuff (#13073), as I see it as very tightly related.

I would ideally like to implement this by next Fri.

jwnimmer-tri, 2020-05-15:
That's not even the right question to ask.  If there have been contexts allocated already but then discarded, the model changes would still be in effect for all usable contexts.

The relevant fact to know is if there are any live contexts, which is exactly the kind of thing you should solve in your application code, not rely on SceneGraph or LeafSystem to try to track for you.

The Anzu planning code dug its own grave with its abuse of widely shared contexts with no abstraction barriers that encapsulate them.  I would rather try to solve this in-house.  I don't think adding a Context GC system into Drake itself should be our first choice.

SeanCurtis-TRI, 2020-05-15:
I'm a bit confused. I thought/hoped `SceneGraph` was clear that *any* modification to the model has *no* effect on pre-existing contexts.

jwnimmer-tri, 2020-05-15:
Right.  And because of that, Eric wants to fail-fast in our application code if we _did_ have a Context hanging around that would not be updated, because they we'd be using incorrect data for future computations. He wants the application-layer invariant that "all of my SG contexts reflect the model modifications made as of point X in my code".

EricCousineau-TRI, 2020-05-15:
@SeanCurtis-TRI Sorry, yeah, your edit clarified. Yes, it won't affect prior contexts.

> He wants the application-layer invariant that "all of my SG contexts reflect the model modifications made as of point X in my code".

@jwnimmer-tri I don't feel like encapsulation / abstraction barriers is really the right requirement on users? Specifically, I *don't* want it to be shoveled into "application-layer" invariant.
e.g. I don't want to use Anzu planning abstractions. I want to use as few layers as possible, and I want to call model-value-geometry-modifying functions on a `SceneGraph`, and have concrete evidence that I'm not foot-gunning myself. Requiring users track this sounds like a backwards requirement?
And it's not "GC" per se, it's just instance counting, no "active" component.

EricCousineau-TRI, 2020-05-15:
However, high-frequency GitHub replies usually indicates that it's best to have a meeting.

Y'all have time on Monday?

jwnimmer-tri, 2020-05-15:
> I don't want to use Anzu planning abstractions. I want to use as few layers as possible, ...

Drake is for solved problems.  Anzu is for working out the right solutions.  In the limit, Drake should provide solutions for all relevant problems, but as we're still figuring out what the right solutions look like, there should be no shame in prototyping the ideas in Anzu until we're happy with them.  As such, never using a shim in Anzu is too high of a standard.  Every shim does indicate a problem in Drake, but that's just how exploration of the problem space works.  Once we like a solution, we will upstream it.

In particular, I dispute that "Count how many Contexts have been allocated" is at all near to an acceptable solution to the correctness challenges you are working on.  Even "Check that all current Contexts have a congruent serial number" would be better.  We should try several of those ideas in Anzu, and then once we have the correct one, we upstream it.

EricCousineau-TRI, 2020-05-15:
> [...] there should be no shame in prototyping the ideas in Anzu until we're happy with them.

I don't see how that counters my point?
And I also don't see how "not using certain Anzu solutions" is too high of a standard. At present, my *prototype* is ensuring I know that absolute minimum for what I want to do, and have a clear and concise way to eliminate hacks.
Whether or not there's time for that is up to me, no one else. Whether or not people are happy with my time allocations are another story.

> [...] is at all near to an acceptable solution [...]

I can buy that. Is there a rewording of the question / issue title that would make you happier?

Once again, I don't feel like we're going to be too productive hashing this out in GitHub replies. I will send a meeting invite. Feel free to accept or reject.

jwnimmer-tri, 2020-05-15:
> I can buy that. Is there a rewording of the question / issue title that would make you happier?

Not that I can think of, no. I don't think this is useful at all as a _Drake_ ticket. Any discussion of a solution relates to the particulars of Anzu's current code and use cases. There is an incredible amount of Anzu code and cross-cutting concerns in play, and the solution relates more to the acute choices in our planning stack than anything fundamental about Drake.

sherm1, 2020-05-15:
I would not like to see Drake Systems keeping track of Contexts in any way. I don't have any architectural objection to mutable counters, serial numbers or whatever in a System if someone wants to know how many times some method has been invoked, though I don't see yet what problem that would solve.

sherm1, 2020-05-17:
I don't see a Drake issue here? Systems and Contexts are intentionally separate objects in Drake. If you modify a System but then continue to use stale Contexts, that's an application problem. If there is a Drake System feature that would be useful to the application for solving this problem (like a counter or serial number), please file an issue. I'll assign this to `team: manipulation` for now for disposition.

EricCousineau-TRI, 2020-05-18:
Per VC with Sean and Jeremy, the need for this feature is a symptom of not having the collisions / collision filters specified in SDFormat (which Calder also alluded to). Closing, and will update relevant Anzu issue.

> I don't see a Drake issue here? [...]

This acutely may not be a Drake issue (now that we've discussed the actual workflow), but it's still a side effect of the fact that we don't really have a great solution to #9691, only mitigations that make it slightly less painful. I don't have any simple solutions in mind, though, so I will leave it where it is.

closed 2020-05-18 by EricCousineau-TRI