RobotLocomotion/drake #13330: doxygen: Possible to have heading anchors / links automatically?

EricCousineau-TRI, 2020-05-16:
I've found it annoying to post links to Doxygen, but then have to write out prose to refer to a section.

For example, [this StackOverflow post](https://stackoverflow.com/questions/61841013/how-to-dampen-multibodyplants-compliant-contact-model-in-a-simulation#comment109380440_61841013), I wanted to link to the some of the subsections showing the math on penalty method point contact model; however, I couldn't get an anchor to it, even if I opened up Firefox's Developer Console, used the inspect, and checked the `<h4/>` element to see if it has an `id=""` attribute or a preceding `<a href="#anchor"/>` tag:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html
> [image]

I've been spoiled by GitHub Markdown and Sphinx, where you get it for free. Example:
> [image]
https://drake.mit.edu/#tutorials

Some possible solutions:

- See if Doxygen has this option?
- On an as-needed basis, add this in manually

*Traceability*: Screen caps taken today, drake @ dd4e089f93

EricCousineau-TRI, 2020-05-16:
Reworded to "show" the anchor.
In fact, there *was* an anchor, but I didn't realize it until I was searching the source (b/c I forgot where the section was):

> https://github.com/RobotLocomotion/drake/blob/dd4e089f93e870e306ed1c34dc4c3ae6a1e0a2ad/multibody/plant/multibody_plant.h#L1282-L1283
> https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#mbp_penalty_method
> [image]

jwnimmer-tri, 2020-05-18:
I'm going to un-assign this from Jamie and Kitware.  I don't think this is a build system ticket.

Eric, if you'd like the class comments to choose different markup (i.e., not `@anchor`) in order to improve the generated output, then open (or reopen & re-title) an issue against those class(es).  (In the cited case, it would be a documentation issue against the Dynamics team.)

If you'd like the Doxygen tool itself to render markup differently by default, then open (or upvote) a ticket against on upstream repo.

I'm not going to prioritize any build system effort to mitigate this.  Our default policy is to work within the bounds of whatever Doxygen provides.

I'll plan to close this later today unless I hear otherwise.

EricCousineau-TRI, 2020-05-18:
SGTM. I've updated the title.

jwnimmer-tri, 2020-05-18:
\CC @sherm1 @amcastro-tri

sherm1, 2020-05-18:
I'm not seeing a Dynamics-area request here. I had interpreted this originally as a request for a general devops improvement. If it is asking for specific code changes, please replace this issue with something narrower. I'm disconnecting this from mbp and dynamics; @EricCousineau-TRI please close or relabel.

jwnimmer-tri, 2020-05-18:
I think the acute request was that `mbp_penalty_method`, `mbp_equations_of_motion`, `model_instances` in MultibodyPlant all use better anchors, which was a dynamics issue.

sherm1, 2020-05-18:
Those would be great topics for an issue!

jwnimmer-tri, 2020-05-20:
Well, this lacks a team label now so it crossed my radar in the daily check of the labels.

@EricCousineau-TRI it sounds to me like Sherm objects to re-purposing the issue to be the MbP docs improvements that it got scoped down to.  So I'll close it now, and ask you to re-file it as a dynamics issue with a differently-phrased top post for better clarity.

closed 2020-05-20 by jwnimmer-tri

EricCousineau-TRI, 2020-05-20:
Sounds good, thanks!