RobotLocomotion/drake #13337: Improve hydroelastic performance

sherm1, 2020-05-18:
Should be able to simulate the simple gripper test
- using simple hydroelastic fingers (e.g., spheres or capsules) against a rigid cylinder
- no more than 20% slower than our best existing point-based approach (currently TAMSI)
- using some Drake integration method or a modified TAMSI.

This is an Epic issue covering several projects, divided between
- speed up hydroelastic computations: #13114 #13130 #13140 #13321
- do fewer hydroelastic computations: #12528, @amcastro-tri's solver work (issue?)

(Extracted from bigger epic #11170, though I relaxed the performance requirement a little.)

amcastro-tri, 2020-05-19:
@SeanCurtis-TRI, do you want me to update the simple gripper to use a hydro model? I could try our newest integration alternatives and evaluate if a new one is needed.

SeanCurtis-TRI, 2020-05-19:
Is there a specific reason we've called out the gripper test? If we go that way, we should be careful about specifying the parameters we're talking about -- it can dynamically alter the number of spheres it uses to approximate the toroidal contact surface. If we convert this to hydroelastic, should we be using a toroidal volume mesh to model the same soft contact surface?  I can go on.

The point of this is to ask have we considered the implications of picking the benchmark?

I would propose that we use the rolling sphere demo instead (at least as the first benchmark in a series of graduated benchmarks). It has the following advantages: 

1. It's already instrumented for point-vs-hydroelastic contact.
2. It's hydro configuration can support combinations of rigid ball/soft ground and soft ball/rigid ground.
2. It represents the most generous scenario for achieving the performance target: it's a ball vs a ground (plane or half space depending on hydroelastic types of the two objects).
    - It eliminates the complexity of tet-tri intersection in favor of tet-plane intersection. (or tri-half space).
    - BVH issues are somewhat reduced because we're not doing BVH-BVH intersection, but BVH-plane or BVH-halfspace.

Certainly, getting the performance of the rolling sphere down does not imply success of more complex scenarios. But, I'd strongly suggest, inability to get the rolling sphere performance to be comparable precludes the ability to get any more interesting scenario performant.

amcastro-tri, 2020-05-19:
That's a good point. One reason not to use the sphere is that it's *too easy*, in that any solver usually does good with it. The gripper is way harder in this regard. I can always modify the gripper example to make the comparison fair (same geometry, etc.)

Notice I am not planning to push this into master (not yet at least), but I'd just play in a dev branch to enable measuring performance in a relevant case. Similar as how @DamrongGuoy put together the bowl-vs-sphere case in a dev branch first.

Another relevant case is "piles of objects", but as we know hydro doesn't support that yet.

SeanCurtis-TRI, 2020-05-19:
"Too easy" is an interesting statement. When I run this on master (on my laptop), Point contact runs the simulation in 0.25 s. Hydroelastic runs in 3.25 s. Given the goal: 

> no more than 20% slower than our best existing point-based approach

hydroelastic should run in no more than 0.3 seconds, and not 3.25 seconds. More than an order of magnitude. So, it doesn't seem to be "too easy" yet.

amcastro-tri, 2020-05-19:
Ah! that's interesting. Are you using continuous integration for both?

SeanCurtis-TRI, 2020-05-19:
Yeah. I just went with the defaults. All I did was change models (and turn off the realtime rate).

If the solution is to simply pick a different integrator based on hydroelastic model and it then runs fast enough, then I would certainly consider the first hurdle cleared and we can move on to the simple gripper. (Of course, we'd want to document which integrator we should pick...or other settings necessary to improve performance.)

sherm1, 2020-05-19:
> Is there a specific reason we've called out the gripper test? 

I transferred this from #11170 without rethinking it. But it does seem like an appropriate metric for us for a number of reasons:
- It models a small but realistic manipulation problem, which is our target application.
- It has problematic slip/stick transitions which TAMSI is particularly good at. That makes finding a viable off-the-shelf solver for hydroelastics difficult and might require a modified TAMSI instead.
- It uses the tri/tet mesh intersections that will be our bread-and-butter 
- Ante has been using the gripper model as his primary integrator benchmark.

I do agree that being able to hit this performance goal with the rolling sphere is a likely prerequisite. That could make a good sub-issue for this Epic. I think we should keep the gripper as a target here as more likely to represent the hurdle we need to jump to make hydroelastics generally useful for manipulation applications.

SeanCurtis-TRI, 2020-05-19:
Agreed. I'd propose that the rolling sphere be targeted first. The simple gripper second. Based purely on my cynical belief that failure to hit the target for the rolling sphere implies failure to hit the target on simple gripper.

sherm1, 2020-05-19:
BTW I think it is _conceivable_ that it might be harder to hit the 20% goal with the rolling sphere than with the gripper -- the rolling sphere might be so small that it does nothing of substance other than computing contact so is disproportionately dominated by the relative cost of finding a point contact vs. creating a contact surface mesh.

SeanCurtis-TRI, 2020-05-19:
Sure. Conceivable. But I still wouldn't bet on it.

xuchenhan-tri, 2022-09-08:
This issue probably deserves an update. We now have discrete hydro and SAP. My question is, is this issue still relevant?

sherm1, 2022-11-16:
With SAP and Discrete Hydro I think this is no longer relevant.

closed 2022-11-16 by sherm1