RobotLocomotion/drake #13379: manipulation station end_effector_teleop_sliders is dead code

jwnimmer-tri, 2020-05-22:
We landed `end_effector_teleop_sliders` (#10632) when Drake still supported python2 and python3, so we had CI coverage of the code in the python2 builds.  Since that point, we've dropped python2 support.  And so, this comment becomes relevant:

> For my part, I [Jeremy] would be okay if the end_effector_teleop_mouse landed on master with a disclaimer that it only works on python2 on Ubuntu + an error message if the user tries to run it on Ubuntu with python3 anyway. In a year(?) from now when we drop support for python2 we'll have to remove the program if nobody steps up to upgrade it to python3, but we can cross that bridge when we come to it. There's every chance someone gets python3-pygame working in Drake in the meantime.

-- https://github.com/RobotLocomotion/drake/pull/10632#issuecomment-463290730

As far as I can tell from a quick look, we still have no pygame in Drake, and its not available in Ubuntu.

I'd like to avoid untested, dead code living on master.  We could imagine stubbing `import pygame` with an inert fill, so that the code still runs but receives no input -- then at least we'd have some statement coverage in the `--test` mode.  However, that doesn't let users actually use it, it just helps mitigate bitrot.

Does macOS offer pygame?  Maybe we could enable for users it there, and have only the test work on Ubuntu.

\CC @RussTedrake @sammy-tri @peteflorence looking for your suggestions here.

peteflorence, 2020-05-22:
Hey Jeremy, pygame is available in python3 from pip now!

jwnimmer-tri, 2020-05-22:
Ah, yes; I see it at https://pypi.org/project/pygame/#files now.  We can see if the whl there works on Ubuntu 18.04.  I also see now that https://packages.ubuntu.com/focal/python3-pygame exists on Focal.  We'll have Focal CI up in a month or two; that could be another solution.

PR #13381 (2020-05-26): workspace: Add dependency on pygame
Closes #13379.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13381)
closed 2020-05-26 by jwnimmer-tri