RobotLocomotion/drake #13387: are the unicorns gone?

EricCousineau-TRI, 2020-05-22:


EricCousineau-TRI, 2020-05-26:
Working on this now, should have solution shortly.

EricCousineau-TRI, 2020-05-26:
Solution posted: #13407

SeanCurtis-TRI, 2020-05-27:
BTW This issue seems to serve for you as a work-tracking ticket. However, it provides very little insight for future developers. Could you extend the original comment to describe the "hurt" as that may better enable other developers to better contribute in this regard. Otherwise, it's all trapped inside your mind.

sherm1, 2020-05-27:
To second Sean's comment, I am worried by this and #13398 that the straightforward abstraction represented by AbstractValue is being compromised unnecessarily. I understand that we may need to restrict our use of that abstraction in some cases to avoid trouble with pybinding, but that should not affect a user who is working only in C++ and might want to make ports or cache entries containing types that wouldn't pybind nicely.

jwnimmer-tri, 2020-05-27:
As shown by the above-linked PR #13407, there is a solution that still permits type-erasing `std::vector<>` by adjusting how the pybind adaptation is formulated.  The plan is **not** to exclude `std::vector<>` as a port or cache entry type.

EricCousineau-TRI, 2020-05-27:
Reworded to focus on bindings, not `Value<>`.

PR #13407 (2020-05-30): py math, multibody: Bind Value[List[T]] for needed types
Resolves #13387

Requires:
- [x] #13408
- [x] #13409
- [x] #13410
- [x] #13405 (un-reverting changes for ports)

\cc @mntan3 @amcastro-tri 

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13407)
closed 2020-05-30 by EricCousineau-TRI