RobotLocomotion/drake #13391: Dropping Ubuntu 18.04 (Bionic) support

EricCousineau-TRI, 2020-05-22:
From Slack: https://drakedevelopers.slack.com/archives/C2NN588UA/p1590171002083800

Tentative date is October 2021. May change.

Marginally relates #13102

jwnimmer-tri, 2021-01-05:
The post above lost some of the nuance from the slack thread.  My sentiment there was a "no earlier than" date, rather than a target date.

In Drake, my general principle is to support the two most recent stable releases for a given platform (Ubuntu, macOS).  For Ubuntu, that means the LTS flavors; e.g., as of today we have 18.04 and 20.04 supported.

Around the time that Ubuntu 22.04 is released in late April 2022, we would want to phase out support for 18.04.  A precise timeline hasn't yet been discussed.  As we start ramping up 22.04 beta CI, we'd ideally like to keep the CI spend in check, so would like to start winding down 18.04.  Whether we drop 18.04 support as of March, April, or May of 2022 is up for discussion.

(All of the above discussion relates to _end user_ support.  Certain developer-oriented features such as code coverage, website generation, or ubsan CI might phase out earlier.)

jamiesnape, 2021-01-05:
Previous tradition, I think, has been that we mostly just keep release builds going after the six month point before 22.04 lands, and I imagine we will keep producing binaries and Docker images right until the cut-off date. Depends on budget and whether we switch Mac to AWS, of course.

jamiesnape, 2021-01-05:
The one difference is that we have "weekly" builds now, so we can keep more tooling or configurations going for longer if we want, or at least for as long as it does not cost too much in developer time (since that dwarfs CI cost).

jwnimmer-tri, 2021-11-11:
In January of 2022, we should announce (in our release notes) that Ubuntu 18.04 support is on its way out.  This would give our users the traditional 3+ months of notice.

jwnimmer-tri, 2021-11-30:
In the v0.37.0 release notes (#16154), I've proposed April 1st, 2022 as the drop date for Bionic.

jwnimmer-tri, 2022-01-05:
Also note that in particular, until https://github.com/googlecolab/colabtools/issues/1880 gets resolved, dropping support for Ubuntu 18 also implies dropping support for Google Colab.  Hopefully Google can upgrade to Ubuntu 20 within the next few months.

jwnimmer-tri, 2022-01-06:
Possibly also worth mentioning directly -- dropping Ubuntu 18 means that our minimum supported Python version will bump up to  3.8 or newer.  That means our wheel builds build drop 3.6 and 3.7 as of April.  I believe that this is consistent with [NEP 29](https://numpy.org/neps/nep-0029-deprecation_policy.html).

jwnimmer-tri, 2022-01-14:
Also worth a shout is #16215 -- related to drake-visualizer.

BetsyMcPhail, 2022-04-12:
In addition to the outstanding changes in #16931 and #16938, Bionic or 18.04 is still mentioned / used in the following places:

- Stable Release instructions in `doc/_pages/from_binary.md` and `doc/_pages/docker.md` (to be removed after next release)

Various packages in tools/workspace:

- `tools/workspace/doxygen`
- `tools/workspace/drake_visualizer`
- `tools/workspace/vtk`
- `tools/workspace/intel_realsense_ros`
- `tools/workspace/pygame_py`
- `tools/workspace/ros_xacro`

The following comments:

- `bindings/pydrake/common/test/deprecation_test.py`:        # For Bionic Python3, the behavior of autocompletion seems to
- `bindings/pydrake/multibody/math_py.cc`: // `symbolic_py.cc`, we must also enable this for clang 9 on Bionic, as is
- `examples/manipulation_station/BUILD.bazel`:    # linux-bionic-clang-bazel-nightly-coverage
- `solvers/test/augmented_lagrangian_test.cc`: // inside the test as a templated lambda, but our current compiler on Bionic
- `solvers/test/augmented_lagrangian_test.cc`: // inside the test as a templated lambda, but our current compiler on Bionic
- `tools/dynamic_analysis/valgrind.supp`: # packaged for Ubuntu Bionic. Fixed in IPOPT 3.12.11 per
- `tools/workspace/uwebsockets/repository.bzl`:        # the GCC7 used on Ubuntu 18.04 (Bionic). Do not upgrade until
- `tools/workspace/uwebsockets/repository.bzl`:        # Bionic support is dropped.
- `doc/images/drake-dragon.README.txt`: # For Ubuntu 18.04.

jwnimmer-tri, 2022-04-12:
>  Stable Release instructions in doc/_pages/from_binary.md and doc/_pages/docker.md (to be removed after next release)

The required changes to the stable release documentation have been added to #16939 now.

BetsyMcPhail, 2022-04-12:
@svenevs is going to take a look at tools/workspace/...

PR #17567 (2022-07-14): Remove various lingering references to 18.04 support
Closes #13391.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17567)
closed 2022-07-14 by EricCousineau-TRI