RobotLocomotion/drake #13393: Why is the default clang version 9 and how do you install it?

cjds, 2020-05-23:
> Before you post:

Using drake externally on master in a bazel project

This change

https://github.com/RobotLocomotion/drake/commit/808d61a49999737523aaeb87a753f3bd5ee16d52

broke our build in this way because we do not have clang format 9 installed. Why is it defaulted to clang format 9 and are there instructions on how to install it?

```
Use --sandbox_debug to see verbose messages from the sandbox
Traceback (most recent call last):
  File "/root/.cache/bazel/_bazel_root/574cc6b7ab86ca2c1b57683005744cf4/sandbox/linux-sandbox/79/execroot/drake_testy_boi/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/tools/vector_gen/lcm_vector_gen.py", line 656, in <module>
    main()
  File "/root/.cache/bazel/_bazel_root/574cc6b7ab86ca2c1b57683005744cf4/sandbox/linux-sandbox/79/execroot/drake_testy_boi/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/tools/vector_gen/lcm_vector_gen.py", line 652, in main
    return generate_all_code(args)
  File "/root/.cache/bazel/_bazel_root/574cc6b7ab86ca2c1b57683005744cf4/sandbox/linux-sandbox/79/execroot/drake_testy_boi/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/tools/vector_gen/lcm_vector_gen.py", line 632, in generate_all_code
    generate_code(src, **kwargs_for_generate)
  File "/root/.cache/bazel/_bazel_root/574cc6b7ab86ca2c1b57683005744cf4/sandbox/linux-sandbox/79/execroot/drake_testy_boi/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/tools/vector_gen/lcm_vector_gen.py", line 591, in generate_code
    [get_clang_format_path(), "--style=" + style, "-i"] + cxx_names)
  File "/root/.cache/bazel/_bazel_root/574cc6b7ab86ca2c1b57683005744cf4/sandbox/linux-sandbox/79/execroot/drake_testy_boi/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/tools/lint/clang_format.py", line 29, in get_clang_format_path
    raise RuntimeError("Could not find required clang-format at " + path)
RuntimeError: Could not find required clang-format at /usr/bin/clang-format-9
----------------
Note: The failure of target @drake//tools/vector_gen:lcm_vector_gen (with exit code 1) may have been caused by the fact that it is running under Python 3 instead of Python 2. Examine the error to determine if that appears to be the problem. Since this target is built in the host configuration, the only way to change its version is to set --host_force_python=PY2, which affects the entire build.

If this error started occurring in Bazel 0.27 and later, it may be because the Python toolchain now enforces that targets analyzed as PY2 and PY3 run under a Python 2 and Python 3 interpreter, respectively. See https://github.com/bazelbuild/bazel/issues/7899 for more information.
----------------
```

My `clang-format` version 6 is located at `/usr/bin/clang-format`

jwnimmer-tri, 2020-05-24:
In the latest release notes at https://drake.mit.edu/release_notes/v0.18.0.html#build-system-and-dependencies we have:

> Use clang-format version 9, not version 6 (#13259, #13260, #13286, #13287)

... which has links to the PRs in question.  In particular #13260 shows the changes to the setup scripts for Ubuntu.

For any builds of Drake from source, as noted in https://drake.mit.edu/from_source.html#mandatory-platform-specific-instructions we expect that users run https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/install_prereqs.sh beforehand, which will install `clang-format-9` in this case.

cjds, 2020-05-24:
Gotcha. I didn't know I had to run the `prereq_install.sh` script regularly if I want to keep on master. Thanks for the info

closed 2020-05-24 by cjds