RobotLocomotion/drake #13396: common: Should teach `Value<>` to only physically store certain types of `Eigen::Matrix<>`

EricCousineau-TRI, 2020-05-25:
Acute C++ issue that feeds into #13388
Splitting off issue from:
https://github.com/RobotLocomotion/drake/issues/13162#issue-608546634
https://github.com/RobotLocomotion/drake/pull/13166#issuecomment-633704111

I am currently struggling between these two solution paths:

1. Teach `Value<T, enable_if_t<is_eigen_type_v<T>>>` to redirect to `Value<SimplifiedEigenType<T>>`. However, `get_value`, `get_mutable_value`, `set_value` may become slower (e.g. requiring copies and checks).
2. Completely disallow `Value<T>` if `is_eigen_type_v<T> && !is_same_v<T, SimplifiedEigenType<T>>`.

I think any solution path is tractable, though may be annoying in terms of meta-programming noise.
(2) is by far simplest, but may be backwards incompatible.
(1) is annoying, but tractable, but may require `const T& get_value()` to become `T GetValue()`.

@jwnimmer-tri @SeanCurtis-TRI Can I ask y'all to weigh in on this?
Ultimately, I think the current usage of `Vectord<N>` can be preserved through `GeometryProperties` and other interfaces, but may require site-specific logic to do so.

jwnimmer-tri, 2020-05-26:
Certainly (1) is a non-starter.  Value has to be **fast**.

SeanCurtis-TRI, 2020-05-27:
I agree with @jwnimmer-tri .

Certainly (2) is viable. And as for sites like `GeometryProperties` that want to opt into faking the storage of fixed size vectors, can we wrap all of that functionality into some utility class that they can simply reuse? (With the caveat that, today, that may be `GeometryProperties` itself with the possibility of hoisting it out of `geometry` if other classes in the future want to do something similar.)

jwnimmer-tri, 2020-05-27:
=> PR #13398 already has some discussion for the solution path.

EricCousineau-TRI, 2020-05-28:
Sorry for not posting integration WIP PR here; the final result could look like:
https://github.com/RobotLocomotion/drake/pull/13160

Specifically, `Value[VectorX[T]]`:
https://github.com/RobotLocomotion/drake/blob/1c6fbbe8969ea3772218b2e45602dab1b4561191/bindings/pydrake/common/test/value_test.py#L100-L120

EricCousineau-TRI, 2020-05-29:
Per discussion in #13398 and #13440, will close this.

Summary is:
Since we can provide a (meaningful) token to represent each vector / matrix type, we shouldn't necessarily prevent; see comment here for what associations are possibly between C++ and Python (without needing concrete `typeid` registration, as is done with `std::vector<T>` and `List[T]`):
https://github.com/robotlocomotion/drake/pull/13440#pullrequestreview-420594093

> If I were to bind the types, then yeah, it would look like either:
> `Vector4 = Vector_[float, 4]`, `VectorX = Vector_[float, Dynamic]`
> or
> `Vector4d = Vector[float, 4]`, `VectorXd = Vector[float, Dynamic]`
> 
> And ideally, I would bind to still work with `AbstractValue.Make`, but explicitly; e.g.:
> ```
> AbstractValue.Make(np.array([1., 2.]))  --> Value[VectorXd]
> AbstractValue.Make(Vector4d([1., 2., 3., 4.]))  -> Value[Vector4d]
> ```
> This would be done by making instantiations of `Vector[]` have special inheritance `np.ndarray`.

---

However, in the case of `GeometryProperties`, the goal should be to use `Rgba` (something like #13441), so currently `Value[Matrix[anything]]` is moot.

closed 2020-05-29 by EricCousineau-TRI