RobotLocomotion/drake #13416: MultibodyPlant set_penetration_allowance pre-finalize

jwnimmer-tri, 2020-05-27:
It's cumbersome to have to wait to call `MultibodyPlant::set_penetration_allowance` until `Finalize` is done.  If I want to tweak plant defaults, I should be able to do that just after I create it, not somewhere far away in the code just before I start to create my simulator.

https://github.com/RobotLocomotion/drake/blob/2b70de6f5114f74dde94f387f3e6591d1f687cb2/multibody/plant/multibody_plant.cc#L1001-L1004

The [method documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a8cdb3408c412b2af94476886dc35cc71) does not warn users about the Finalize pothole.

Related to #13064 somewhat.

sherm1, 2020-05-27:
Sounds like a good idea, but FYI this is currently being reworked per #13064 #13289 #13371 to allow per-geometry parameters.

jwnimmer-tri, 2020-05-28:
Hopefully those are also configurable pre-Finalize!

But for the record, in the meantime @amcastro-tri has said in slack that he can do this change sooner than that rework.

amcastro-tri, 2020-05-29:
Resolved in #13435.

closed 2020-05-29 by amcastro-tri