RobotLocomotion/drake #13446: visualization: Director

EricCousineau-TRI, 2020-05-29:
## Problem / Repro

See Slack convo for more details:
https://drakedevelopers.slack.com/archives/C43KX47A9/p1590702219060700

Andy's report:

> I have occasionally seen a model in drake_visualizer spontaneously "erode". I think I have finally been able to reproduce it (although I don't yet have a minimal working example ready). However, If you run a simulation that has one static body (welded to world) with visual geometry and a second free-body (say a box with 1kg mass) that is simply dropped from z=0 off to z=-inf, after about a minute of sim time I see the first body "erode" in the visualizer.  This is on v0.18.

Repro code:
https://github.com/EricCousineau-TRI/repro/commit/b2d7f668ed1b5a757b984c54845b78a56e991d5b

> [image]

## Root Cause

@patmarion's answer:
> hi!  right, the visualizer is automatically adjusting the near & far clipping planes so that all geometry is within bounds.  Normally this is a useful feature, but if your object is falling to infinity while another object is at the origin then you will have issues with depth buffer resolution.
https://www.khronos.org/opengl/wiki/Depth_Buffer_Precision
A fix would be to insert some python code to set fixed near/far clipping bounds.  Then the object falling to infinity will eventually clip away and the rest of the scene is unaffected.  I can suggest some code if that's a solution you want to try.

## Solution

From Pat's suggestion:

```
def on_render(render_window, event):
    near, far = view.camera().GetClippingRange()
    view.camera().SetClippingRange(near, min(far, 1e4))
view.renderWindow().AddObserver('StartEvent', on_render)
```

\cc @SeanCurtis-TRI

EricCousineau-TRI, 2020-05-29:
Copying from PR review: https://github.com/RobotLocomotion/drake/pull/13447#pullrequestreview-420998914

> I tried Pat's advice verbatim, found that it did not fully work, and Andy had mentioned making this adjustment:
> 
> > Thanks Pat, I made a small modification - similarly setting the near clipping range to min(near, 1) and that seemed to eliminate the erosion on my end - Thank you!
> 
> I found the `near = 1` was too large, because if I zoomed in it clipped the block.

SeanCurtis-TRI, 2020-05-29:
For the record, "falling away to -inf" is merely one special case. More generally, as the bounding box of the scene elements increases in volume, that leads to "eroding" clipping planes.

PR #13447 (2020-05-30): drake_visualizer: Prevent clipping volume from going to infinity
Resolves #13446

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13447)
closed 2020-05-30 by EricCousineau-TRI