RobotLocomotion/drake #13457: Accelerometer and Gyroscope sensors

mposa, 2020-05-31:
Since attic/ is being deprecated, we will need MBP-friendly implementations of IMU sensors. It's on our todo list, unless someone wants to tackle it first.

sherm1, 2020-05-31:
Do you want just ground truth spatial acceleration results, or a real device simulation?

sherm1, 2020-06-03:
MBP can already provide SpatialAcceleration for each body via an output port (we could do it by API also but I'm not sure that's wired up yet). If there is a body representing the IMU that would give a ground truth reading. Is that usable?

mposa, 2020-06-03:
That's definitely the data that's needed (spatial acceleration and angular velocity), though we'll need to wrap it into objects "IMU/Accelerometer/Gryo" that only provides the actual measurements. I think the basic question is whether or not these sensor systems exist in Drake or only in our dairlib ecosystem.

sherm1, 2020-06-03:
Ah, yes it would be nice to have pre-built IMU sensors living in `drake/systems/sensors` along with rotary encoders, rgbd, etc.

mposa, 2020-07-05:
Resolved via https://github.com/RobotLocomotion/drake/pull/13501 and https://github.com/RobotLocomotion/drake/pull/13604 and

closed 2020-07-05 by mposa