RobotLocomotion/drake #13459: mbp: Can MultibodyPlant::CalcAppliedForces be public?

EricCousineau-TRI, 2020-05-31:
Per @mposa's request on Slack:
https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1590954111154700

Looking at the interface, seems pretty reasonable?
Will make a quick PR. May punt to another person if it becomes nontrivial ;)

amcastro-tri, 2020-06-01:
Thanks @EricCousineau-TRI and @mposa.
The discussion on Slack doesn't say (at least to me) why the already available `MBP::CalcForceElementsContribution()` doesn't solve the particular problem. @mposa, could you expand on this? or @EricCousineau-TRI if you happen to have had that conversation.

EricCousineau-TRI, 2020-06-03:
Closed per discussion here:
https://github.com/RobotLocomotion/drake/issues/13476#issuecomment-638446799

closed 2020-06-03 by EricCousineau-TRI