RobotLocomotion/drake #13465: Design of input (not parameter) for no-op IIWA command

jwnimmer-tri, 2020-06-01:
From #13340:

@siyuanfeng-tri writes:
> sorry i didn't catch this earlier.
> i just realized that setting desired q to measured q is probably not the best thing to do before receiving a real command... since that effectively wipes out the position feed back loop. this would be more weird when we compute the desired velocity based on numeric difference between commanded q, and would be extra weird if we specified a integral gain in the pid loop downstream. we are currently doing all of these in the standard iiwa pipeline..

jwnimmer-tri, 2020-06-01:
I'll admit I don't follow the concern; @sammy-tri do you understand?

sammy-tri, 2020-06-02:
Maybe...  I believe the concern is what the `InverseDynamicsController` will do if an external force acts on the arm before the first command is received?

siyuanfeng-tri, 2020-06-02:
let me try again after the meetings... but can you check this really quick? just let a iiwa sim (start from q = 0s) running for a while without giving it a lcm command message. i think the arm will eventually collapse into a weird configuration, and if you open up iiwa status, the commanded q is chasing the measured q.

sammy-tri, 2020-06-02:
The robot definitely moves when started in a non-zero configuration and left with no command being sent...   but not much.  After 10 minutes of sim time the highest joint velocity I've seen is somewhere around 3e-5, and the joint which has moved the most is about 0.015 radians from it's starting position.  I suspect if an external force were applied this would be more significant.

So I tried dropping a plate onto the iiwa from 5m above it in the simulation.  It pushes the robot quite a bit, but it seems to bounce into a mostly stable configuration...   eventually my sim crashes with an assertion in `MultibodyPlant`, but given that I've got a plate accelerating downward at 1g heading out to infinity, maybe that's a bigger problem (indeed, dropping the plate so that it bounces into the dishwasher seems to avoid this error).

I don't think the behavior I'm seeing is going to have a significant effect on our existing iiwa use cases, but it is different than what `set_initial_position` provided.

jwnimmer-tri, 2020-06-02:
I guess whether the movement is small or large, it's still different than what the `kuka_driver` + control cabinet does, then?  Where does the state live in real life live -- in the `kuka_driver` check of `utime == -1`?  It sounds like we should model that in sim as well.  Luckily we don't need an API change -- we can have the receiver latch the initial position until a transmitted command comes in.

sammy-tri, 2020-06-02:
That's effectively what the driver does in real life.  When FRI enters the "command" phase (which is really the driver's first chance to do anything) it latches the current position and continues commanding that until the `utime == -1` check is no longer true.

siyuanfeng-tri, 2020-06-03:
> we can have the receiver latch the initial position until a transmitted command comes in. 

I like this much better. I think this is equivalent as as automatic `set_initial_postiion()` call with the measured q at the first tick. The old sim behavior as well as kuka_driver has been maintain the first measured position until a real command is given. The new behavior is effectively maintain zero acceleration. So it's kinda like kuka's low impedance mode demo that you can push the arm around effortlessly. But it will keep moving once the arm gets going, so no damping.

The movement difference in sim depends on how well the controller model matches the model in the simulator.

jwnimmer-tri, 2020-06-04:
@siyuanfeng-tri Do you have a WIP PR underway for this?  I could take it over, or start my own afresh if you don't have one yet.

siyuanfeng-tri, 2020-06-04:
yeah i do. i don't like it though.

siyuanfeng-tri, 2020-06-11:
https://github.com/jwnimmer-tri/drake/blob/cmd_recv_latch/systems/framework/leaf_system.h#L79
why isn't this just return owned_model; ?

jwnimmer-tri, 2020-06-11:
Yup, it should have.  I've opened #13540 now to make the WIP branch more visible.

PR #13540 (2020-06-12): IiwaCommandReceiver latches initial position until first command
Closes #13465.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13540)
closed 2020-06-12 by jwnimmer-tri