RobotLocomotion/drake #13475: Fetching drake-apt repository fails due to expired root certificate issue

jamiesnape, 2020-06-02:
```
 https://drake-apt.csail.mit.edu/drake.pub.gpg
 Resolving drake-apt.csail.mit.edu (drake-apt.csail.mit.edu)... Warning: apt-key output should not be parsed (stdout is not a terminal)
 2600:9000:20aa:7e00:15:1bde:5f00:93a1, 2600:9000:20aa:5a00:15:1bde:5f00:93a1, 2600:9000:20aa:3000:15:1bde:5f00:93a1, ...
 Connecting to drake-apt.csail.mit.edu (drake-apt.csail.mit.edu)|2600:9000:20aa:7e00:15:1bde:5f00:93a1|:443... connected.
 HTTP request sent, awaiting response... 200 OK
 Length: 3134 (3.1K)
 Saving to: ‘STDOUT’
 
      0K ...                                                   100%  104K=0.03s
 
 (104 KB/s) - written to stdout [3134/3134]
 
 OK
 Hit:1 http://us-east-1.ec2.archive.ubuntu.com/ubuntu bionic InRelease
 Hit:2 http://us-east-1.ec2.archive.ubuntu.com/ubuntu bionic-updates InRelease
 Hit:3 http://us-east-1.ec2.archive.ubuntu.com/ubuntu bionic-backports InRelease
 Ign:4 https://drake-apt.csail.mit.edu/bionic bionic InRelease
 Err:5 https://drake-apt.csail.mit.edu/bionic bionic Release
   Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 2600:9000:20aa:7e00:15:1bde:5f00:93a1 443]
 Hit:6 https://apt.kitware.com/ubuntu bionic InRelease
 Hit:7 http://security.ubuntu.com/ubuntu bionic-security InRelease
 Reading package lists...
 E: The repository 'https://drake-apt.csail.mit.edu/bionic bionic Release' does not have a Release file.
 The Drake source distribution prerequisite setup script has experienced an error on line 9 while running the command source "${BASH_SOURCE%/*}/source_distribution/install_prereqs.sh"
```

Due to https://support.sectigo.com/articles/Knowledge/Sectigo-AddTrust-External-CA-Root-Expiring-May-30-2020. 

The root cause also impacts the following:

* https://drake-cdash.csail.mit.edu
* https://drake-girder.csail.mit.edu
* https://drake-homebrew.csail.mit.edu
* https://drake-jenkins.csail.mit.edu
* https://drake-mirror.csail.mit.edu
* https://drake-packages.csail.mit.edu

Fixed by infrastructure updates and/or mitigated for Ubuntu distributions by https://usn.ubuntu.com/4377-1/.

Ubuntu 18.04 (Bionic) users should update as follows:
```
sudo rm -f /etc/apt/source.d/drake.list
sudo apt-get update
sudo apt-get install ca-certificates  # upgrade to 20190110~18.04.1
sudo ./setup/ubuntu/install_prereqs.sh
```

Ubuntu 20.04 (Focal) users may wish to upgrade to `ca-certificates` version `20190110ubuntu1.1` since other websites may rely on the expired root certificate, but it is not required for Drake at present.

jwnimmer-tri, 2020-06-09:
To smooth the upgrade path, we need to special-case the upgrade script to check the version of `ca-certificates` and automatically implement the above work-around.

jamiesnape, 2020-06-09:
The correct solution is to get the certificate reissued. I will work on it.

jwnimmer-tri, 2020-06-09:
SGTM.

jamiesnape, 2020-06-09:
Temporary workaround in the script would be something like
```
apt-get update || apt-get install ca-certificates
```
You don't really need to check versions.

jamiesnape, 2020-06-09:
New certificate installed.

closed 2020-06-09 by jamiesnape