RobotLocomotion/drake #13476: Force element contributions

mposa, 2020-06-02:
There is currently no easy way to calculate the contributions of all force elements as a generalized force (net force in generalized velocity coordinates). 

`plant.CalcForceElementsContribution` seems closest to this use case, but it separates out what I think of as generalized forces into two terms:

- generalized forces (for a some forces)
- body forces (for others, including gravity)

Per a conversation with @amcastro-tri , what seems to be missing is the ability to add these two terms together.

This stems from our basic need to calculate the manipulator equations, e..g
```
M(q) * vdot = f + Bu
```
or
```
M(q) * vdot + C(q,v)v = f + Bu
```
where we can reliability compute `f` in this expression.

amcastro-tri, 2020-06-02:
Let me add that the confusion stems from the fact that `CalcForceElementsContribution()` stores the result in a `MultibodyForces`. Unfortunately it wasn't clear for @mposa that the raw result as obtained with `MultibodyForces::generalized_forces()` is not what he wanted. This is clearly a defect to be addressed. 

IMO the problems are:
- Users should not call `MultibodyForces::generalized_forces()` nor `MultibodyForces::body_forces()` but operate on `MultibodyForces` as an abstraction to all possibly applied external forces. 
- There should be a sugar API like `MBP::CalcForceElementsGeneralizedForces()` that would return exactly what @mposa wants.
- A `drake/example` (or tutorial) explaining users how to work with `ForceElements` and/or `MultibodyForces`. 

Very closely related to previous issues #12911 and #12942.

EricCousineau-TRI, 2020-06-03:
Awesome!

However, I would vote against having sugar API in `MBP`, and instead vote for having something directly on `MultibodyForces` that more clearly shows what is wanted, e.g. `MultibodyForces::CalcTotalGeneralizedForces`.

Perhaps `::generalized_forces()` should be deprecated in lieu of a better name, like `::purely_computed_generalized_forces()` or something that shows that it doesn't incorporate all of the force contributions that may be expected.

EricCousineau-TRI, 2020-06-03:
That being said, is #13459 (and it's PR) moot now?
Should I close it and the PR if all the math is available, but just needs some polishing?
That, or I am happy to still port over part of the unittest or whatever.

amcastro-tri, 2020-06-03:
I believe that `CalcAppliedForces` _could_ be useful. It just didn't directly address Michael's need (all generalized forces).
I believe what he needs is more like a `MBP::CalcForceElementsGeneralizedForces()`, `f` in his issue post.

Actually, something like `CalcAppliedForces()` would only add confusion as far as I could tell, given we don't really explain well what `MultibodyForces::generalized_forces()` actually mean. 

So ok, talking myself out of it. I would not make `CalcAppliedForces()`, not yet at least until we provide properly documented sugar to deal with `MultibodyForces` objects.

EricCousineau-TRI, 2020-06-03:
SGTM! Going to close issue and PR for now; can reopen later!