RobotLocomotion/drake #13480: Update Coulomb friction plot label

amcastro-tri, 2020-06-02:
A (late) follow up to #13463. 

I just realized we must update the label of the x axis [here](https://github.com/RobotLocomotion/drake/blob/a58cf34fbcad1c6f7fc43a42289a0b6775bdcaf4/multibody/plant/images/ideal_stiction.py#L22).

Where it says "Pushing Force (N)", it should say "Sliding velocity (m/s)". 

Nit, I'd also advocate for square brackets to place units.

EricCousineau-TRI, 2020-06-02:
Simple issues should really have one owner, otherwise accountability is super unclear.
Taking myself off for now ;)

Also, I think in the time it took you to write this issue, you could have written the PR + motivation yourself ;)

SeanCurtis-TRI, 2020-06-02:
And given where the code lives, this really feels like you should be the owner.

EricCousineau-TRI, 2020-06-02:
Went ahead and did it anywho. It took me 30sec.
@amcastro-tri Would really love it if in the future you made the smaller changes yourself ;)

closed 2020-06-02 by EricCousineau-TRI

amcastro-tri, 2020-06-03:
sry, I was absolute swamped and when I considered it I realized I had no idea what the process to update those png's is, while you guys do. So no, it would've taken me more than 30 secs.

amcastro-tri, 2020-06-03:
In retrospect, I realized I should've assigned this to self and use it as a reminder to work on it later. Sorry if this caused you a distraction @EricCousineau-TRI, @SeanCurtis-TRI

PR #13485 (2020-06-02): plant/images: Update Coulomb friction plot label
Resolves #13480

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13485)