RobotLocomotion/drake #13493: scene_graph: Can haz SceneGraphInspector::GetGeometries(GeometrySet) or something like that?

EricCousineau-TRI, 2020-06-03:
Hoisting Anzu discussion to Drake issue:

From subgraph PR in Anzu:

PR code is based on this bit: 
https://github.com/EricCousineau-TRI/repro/blob/54494a5c5154f19e693e4862fbaa79cddcd78d6f/drake_stuff/multibody_plant_prototypes/multibody_extras.py#L91-L100

> ```py
> def get_geometries(plant, scene_graph, bodies):
>     """Returns all GeometryId's attached to bodies. Assumes corresponding
>     FrameId's have been added."""
>     geometry_ids = []
>     inspector = scene_graph.model_inspector()
>     for geometry_id in inspector.GetAllGeometryIds():
>         body = plant.GetBodyFromFrameId(inspector.GetFrameId(geometry_id))
>         if body in bodies:
>             geometry_ids.append(geometry_id)
>     return sorted(geometry_ids, key=lambda x: x.get_value())
> ```
> 
@SeanCurtis-TRI's comment:
> BTW Blech! I'm sorry that missing API forces you to do this backwards! You should have:
> 
> ```py
> for body in bodies:
>     frame_id = plant.GetBodyFromFrameId(body.get_index())
>     assert(frame_id is not None)
>     geometry_ids += inspector.GetGeometries(frame_id):
> ```

SeanCurtis-TRI, 2020-06-03:
The issue summary doesn't match the code snippets in the issue description. There is no mention of `GeometrySet` in the discussion.  (And thanks for creating the issue. Feel free to assign it to me unless you *really* intend to do it any time soon.)

EricCousineau-TRI, 2020-06-03:
Sounds good on all accounts! Updated title, added the mention of `GeometrySet` in discussion transcript, and reassigned!

SeanCurtis-TRI, 2020-09-25:
@EricCousineau-TRI The main core of this PR has been resolved by #13999, I believe. 

I vote we close this issue and split out the second part into its own issue. Thoughts?

SeanCurtis-TRI, 2020-09-25:
Alternatively, the second part would be *trivial* to provide, so I might just knock it out to close this issue.

EricCousineau-TRI, 2020-09-25:
I'm fine either way!

SeanCurtis-TRI, 2021-04-06:
This issue is resolved in two parts: #13999 gave the ability to collect geometries on a per-frame basis and #14871 does the same thing on a per `GeometrySet` basis.

PR #14871 (2021-04-07): Add ability to convert GeometrySet into unordered_set<GeometryId>
The ability to convert also includes the ability to filter based on role. This includes the public-facing inspector API as well as the GeometryState implementation.

closes #13493

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14871)
closed 2021-04-07 by sammy-tri