RobotLocomotion/drake #13499: mkdoc: Should split out Doxygen -> Sphinx (Napoleon) and test separately

EricCousineau-TRI, 2020-06-04:
Currently, we have our mega-function:
https://github.com/RobotLocomotion/drake/blob/76956e194f197f9ce69236d88aab4e996b24c90a/third_party/com_github_pybind_pybind11/mkdoc.py#L227-L654

That is tested Christmas-tree style:
https://github.com/RobotLocomotion/drake/blob/76956e194f197f9ce69236d88aab4e996b24c90a/tools/workspace/pybind11/test/sample_header_documentation.expected.h

Tasks:

- [ ] We should separate out the docstring processing to its own module (since it's relatively modular), and also have focused tests.
- [ ] As part of this, it would be nice to also simplify `sample_header.h`.

\cc @jamiesnape

EricCousineau-TRI, 2020-06-04:
@BetsyMcPhail Might this be feasible w/in the next 4 weeks? (June 25)

BetsyMcPhail, 2020-06-08:
> @BetsyMcPhail Might this be feasible w/in the next 4 weeks? (June 25)

Sounds good to me.

PR #13651 (2020-07-10): Split mkdoc comment processing to its own module
Move code from sample_header.h to focused tests.

Resolves #13499

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13651)
commit 04396b46 (2020-07-10): Split mkdoc comment processing to its own module (#13651)

* Split mkdoc comment processing to its own module

Move code from sample_header.h to focused tests.

Resolves #13499
closed 2020-07-10 by EricCousineau-TRI