RobotLocomotion/drake #13518: Need a tutorial showing the use of AbstractValue in Input/OutputPorts

RussTedrake, 2020-06-09:
Per the developers meeting today.  Could be added easily to the existing `dynamical_systems.ipynb`. 

cc @mposa

jwnimmer-tri, 2020-06-09:
Looking at #13501, an example of tricky syntax is:
```c++
  // In the constructor ...
  body_poses_input_port_ = &this->DeclareAbstractInputPort(
      "body_poses", Value<std::vector<RigidTransform<T>>>());
```
or
```c++
  // In a Calc method ...
  const auto& X_WB = get_body_poses_input_port()
                          .template Eval<std::vector<RigidTransform<T>>>(
                              context)[body_index_];
```

In particular:
- It is a scalar-dependent abstract port, so needs `template` a keyword.
- It is a `std::vector` of values not just a single value.

These are going to have fairly different spellings in Python vs C++.  Probably both examples are necessary.

jwnimmer-tri, 2024-11-01:
Our `authoring_leaf_systems.ipynb` has a section showing "Abstract-valued Ports".

closed 2024-11-01 by jwnimmer-tri