RobotLocomotion/drake #13519: http://drake-apt.csail.mit.edu/bionic certificate validation failed (expired certificate)

RussTedrake, 2020-06-09:
I cannot run `install_prereqs` on bionic this morning and cannot build `master` without it.  
```
Err:10 https://drake-apt.csail.mit.edu/bionic bionic Release                   
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 13.225.222.28 443]
Hit:11 http://ppa.launchpad.net/nilarimogard/webupd8/ubuntu bionic InRelease   
Get:12 http://packages.ros.org/ros/ubuntu bionic InRelease [4,669 B]     
Err:6 https://storage.googleapis.com/bazel-apt stable InRelease 
  The following signatures were invalid: EXPKEYSIG 3D5919B448457EE0 Bazel Developer (Bazel APT repository key) <bazel-dev@googlegroups.com>
Err:12 http://packages.ros.org/ros/ubuntu bionic InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Reading package lists... Done 
E: The repository 'https://drake-apt.csail.mit.edu/bionic bionic Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://storage.googleapis.com/bazel-apt stable InRelease: The following signatures were invalid: EXPKEYSIG 3D5919B448457EE0 Bazel Developer (Bazel APT repository key) <bazel-dev@googlegroups.com>
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
The Drake source distribution prerequisite setup script has experienced an error on line 9 while running the command source "${BASH_SOURCE%/*}/binary_distribution/install_prereqs.sh"
```

soonho-tri, 2020-06-09:
I think https://github.com/bazelbuild/bazel/issues/5274#issuecomment-458931341 can solve the bazel-apt problem?

soonho-tri, 2020-06-09:
For ROS, please check if https://github.com/ros2/ros2/issues/742#issuecomment-509685308 helps.

jwnimmer-tri, 2020-06-09:
The lines
```
Err:10 https://drake-apt.csail.mit.edu/bionic bionic Release                   
Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 13.225.222.28 443]
```
are due to #13475.  Pinning the issue atop the issues list was not enough; I'll reopen it to ask that the install script self-repair the `ca-certificates` if its is too old.

For the others, perhaps with the apparently increasing prevalence of broken PPAs, we should adjust install_prereqs to print a specific error for this case: "Your computer is broken, Drake needs `apt-get update` to pass before you may install Drake's prereqs, this is not a Drake issue".

PR #16086 (2021-11-12): [setup] Display a disclaimer when 'apt-get update' is broken
Closes #13519.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16086)
closed 2021-11-12 by jwnimmer-tri