RobotLocomotion/drake #13520: WeldJoint::set_X_PC after finalize

ggould-tri, 2020-06-09:
TRI Project Anzu would find it helpful if WeldJoint had (1) a setter for X_PC, which (2) could be called after finalize.  This would greatly simplify our model error system by allowing all error/noise to be applied at runtime.

/cc @joemasterjohn @amcastro-tri @jwnimmer-tri 

Discussion on internal TRI mailing lists leading to this:

> 
> Alejandro --
> 
> It came up in our synthesis stand-up today that we could greatly simplify our code if WeldJoint had a method to set X_PC after finalize.  Do you have a sense of how hard/impossible this would be?
> 
>  --Grant
> 
> Alejandro Castro
> 
> I haven't looked into it specifically. But this quarter Joe started the work on MBP's parameters.
> He started with contact parameters. We'll continue with mass (and maybe rotational inertias). What we learn from exposing those as parameters will definitely help us on how to do the same with things like X_PC.
> Now, Jeremy mentions that this might be pre-context creation though?
> 
> A.
> 
> 
> Jeremy Nimmer
> 
> Sounds like the answer is "not scary impossible" in which case maybe Grant can you open a Drake issue feature request on this?  And CC to Joe on the issue, since it might dovetail with a current project.
>

amcastro-tri, 2020-06-09:
Thanks @ggould-tri. I believe we'd need something very similar to what @joemasterjohn did in #13105 for joints' default positions. 
In this case `WeldJoint` does store the default parameter value for `X_PC`. We'd then need to introduce a method `WeldJoint::set_default_X_PC(const RigidTransfrom<double>& X_PC)`  and make sure this method updates the underlying mobilizer model when called.
Would that do it?

sherm1, 2020-06-10:
FYI See [this comment](https://github.com/RobotLocomotion/drake/issues/13289#issuecomment-641653390) in #13289. (Permitting a change to X_PC would leave all existing Contexts in an invalid state.)

jwnimmer-tri, 2020-07-23:
@ggould-tri I wonder -- what if WeldJoint took X_PC as symbolic instead of double, so we could pass in randomness to be sampled during CreateRandomContext?  That would make the flow even smoother, perhaps?

jwnimmer-tri, 2024-11-01:
There is already an available solution for this, so closing as "not planned".

A weld joint can attach to a FixedOffsetFrame, and the offset frame's offset can already be configured.  That's good enough until we have a sample use case where the offset frame cannot meet the need.

closed 2024-11-01 by jwnimmer-tri