RobotLocomotion/drake #13524: Building and Using pydrake from source

carismoses, 2020-06-09:
I am simply trying to build and use pydrake from source. I have seen the Jupyter notebooks on how to use pydrake and they are helpful, but I would like to do this from source on my machine. My long term goal is to use pybind11 to wrap some existing code I have that interfaces with drake, so that I can use it in python. To this end I am first trying to understand how pydrake acts as a wrapper, and replicate that functionality. However, I am unable to find instructions anywhere on how to build pydrake and use it from source. I have tried `bazel build bindings/pydrake` and then run `python; import pydrake` from the drake/bindings directory, but I get the error `ModuleNotFoundError: No module named 'pydrake.common._module_py'`. What is the usual workflow for building and then using the python bindings?

sherm1, 2020-06-09:
Hi, @carismoses. Thanks for your interest in Drake! Would you mind reposting this in Stackoverflow instead (with a "drake" tag)? That way your question and the answer will become part of the searchable Drake knowledge base. The [Getting Help](https://drake.mit.edu/getting_help.html) page has a link.

closed 2020-06-09 by sherm1

carismoses, 2020-06-09:
No problem @sherm1, just posted.

sherm1, 2020-06-09:
Thanks, Caris!