RobotLocomotion/drake #13526: cmake: Add option to disable building / installing Python bindings

EricCousineau-TRI, 2020-06-09:
Per Slack request from Sam Z:
https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1591737731088300

EricCousineau-TRI, 2020-06-09:
Plan of attack:
- Try quick source-tree hack that does not install bindings
- See if we can formalize this

EricCousineau-TRI, 2020-06-09:
@samzapo Here's the one-liner hack:
https://github.com/EricCousineau-TRI/drake/commit/053a435e4a3d414090848727a55039eaab28d118

I'm briefly checking on my side.

Can you see if this works for you?

samzapo, 2020-06-09:
Yes, that fix worked for me.

EricCousineau-TRI, 2020-06-09:
Just to check, how are you consuming Drake from CMake?

The two options (from https://github.com/RobotLocomotion/drake-external-examples/tree/master/drake_cmake_external)
- `drake_cmake_installed` - Installing via Bazel to somewhere like `/opt/drake`
- `drake_cmake_external` - Using CMake `ExternalProject_Add` to consume Drake from source tree

jamiesnape, 2020-06-09:
Should be easy to formalize, I think. You can assign to me, if you like.

EricCousineau-TRI, 2020-06-09:
Done - thanks!

Only question in my mind is testing - I know we have an in-tree install test for the `drake_cmake_installed` worfklow, but nothing for the `drake_cmake_external` workflow (which makes total sense, come to think of it :P)

samzapo, 2020-06-09:
I'm using the `drake_cmake_installed` pattern.

jwnimmer-tri, 2022-05-06:
Resolving as a duplicate of #13526, which will attack the full problem.

closed 2022-05-06 by jwnimmer-tri