RobotLocomotion/drake #13527: Panda model has incorrect collision geometry

calderpg-tri, 2020-06-09:
The Panda models in drake have incorrect collision geometry on nearly all links. A fixed version is now in anzu, and these changes need to be brought back to drake.

I've nominally assigned myself, but I may not have time to get to this quickly.

cc @sammy-tri

huihuaTRI, 2020-06-11:
I will take care of this one. Could you point (email/slack) me the fixed version?

calderpg-tri, 2020-06-11:
Link to fixed model sent on slack, you'll need to change both [panda_arm.urdf](https://github.com/RobotLocomotion/drake/blob/master/manipulation/models/franka_description/urdf/panda_arm.urdf) and [panda_arm_hand.urdf](https://github.com/RobotLocomotion/drake/blob/master/manipulation/models/franka_description/urdf/panda_arm_hand.urdf).

huihuaTRI, 2020-06-12:
[PR13550](https://github.com/RobotLocomotion/drake/pull/13550) landed.

closed 2020-06-12 by huihuaTRI