RobotLocomotion/drake #13539: Re-org and rewrite of distribution, installation, and build docs

jwnimmer-tri, 2020-06-11:
There are numerous known issue and usability hurdles with Drake's documentation for how users consume Drake releases, nightlies, source builds, pydrake, etc.

It will be most efficient to take a big swing at all of these at once in a single editing pass, because some of the problems relate to how the information is organized, rather than the specific facts being presented.

I will link all of the relevant issues here.

My currently plan is to begin this initiative this July.

jwnimmer-tri, 2020-06-25:
We'll kick this off on week of Monday 7/13.

jamiesnape, 2020-09-08:
I would add the following (feel free to merge up top):

- [ ] #14029 drake-lcm-spy gives NoClassDefFoundError in Docker
- [ ] #13924 cart-pole example throws exception in cmake project

jwnimmer-tri, 2025-01-25:
This is essentially done now.  I've filed #22530 to kick off a last look, but most/all of the acute problems noted above have been resolved, at least in part.

closed 2025-01-25 by jwnimmer-tri